bw1129/PIDtoolbox #2: macOS / Linux binary builds

fujin, 2019-02-06:
I see that the MATLAB Runtime environment is available for 64-bit Linux and macOS. Would it be possible to have built binaries for those or is there some other requirement/pre-requisite work?

bw1129, 2019-02-06:
It’s certainly possible. I just don’t have a copy of matlab for Mac or Linux, but if someone does, it’d be awesome if they built a version. I suppose the directory assignment might be a little different, but certainly no major coding changes. I’ll see if the university will give me a copy.

haplm, 2019-03-20:
I have eval license on MacOS that is still good for like 4 days. So if you tell me how to build it, I will ;-)

haplm, 2019-03-20:
Also - I have installed Octave on MacOS, so if there's a way how to run it under Octave, I can try

bw1129, 2019-03-20:
> Also - I have installed Octave on MacOS, so if there's a way how to run it under Octave, I can try

hey thanks, that's kind of you. I have been trying to port to Octave. Unfortunately it wont run as is. there are quite a few low level differences between the two that you have to overcome with a different programming approach. One thing that's been killing me is Octave graphics are slow. There is some ongoing octave-opengl issue that most wouldnt notice because people dont typically plot a vector that's like >100,000 points long, but that's typical length of gyro trace for example (could downsample but crappy when you zoom on the trace). Anyway, the matlab version is faster now and can import bbl/bfl files directly. Will try porting after the current revision.

haplm, 2019-03-20:
I see.

Anyway, I have checked that I can start it under full Matlab on MacOS. I'm just installing the compiler module so I can compile standalone version. If you point me to towards how to perform the actual compilation it would be great, otherwise I will somehow figure out myself, I guess.