bw1129/PIDtoolbox #3: Fails to load file...

kmitchel, 2019-02-07:
My BB logs will not open...a dong sound plays, and the progress bar freezes.

https://www.dropbox.com/s/f6nkctkybrz2npq/BTFL_BLACKBOX_LOG_iCr4sh_20190205_180121.BBL%20%282%29.csv?dl=0

bw1129, 2019-02-07:
make sure the csv was created from bloacbox explorer. it will not run csv files created from blackbox_decode. second thing to try, place csv in folder with PIDtoolbox.exe.

kmitchel, 2019-02-07:
Moving the csv files into the same directory as the executable has helped. I still have one file that still freezes.  I'll continue to try others.  Should it still load with the wrong debug mode? TY

bw1129, 2019-02-07:
It should be fine now. Just make sure the csv was generated in explorer. That’s critical. Make sure debug set to gyro_scaled, though it should run fine with any debug mode, but PIDtoolbox expects debug variables to contain unfiltered gyro.

DusKing1, 2019-02-11:
I now have the same issue. Two csv files was exported by blackbox explorer, and they are in same folder with .m files. The CMD in matlab said:

> 出错 PTprocess (line 126)
> dat_B=dataB.DataMain(Time_B>ttb(find(ttb>y(1),1)) & Time_B<ttb(find(ttb>y(2),1)),:);
> 计算 UIControl Callback 时出错。

 (I'm using Chinese, and don't know how to change the language. Sorry for the mess)

DusKing1, 2019-02-11:
OK, after restarting my computer some problem gone. But still got same frozen situation as kmitchel.

bw1129, 2019-02-11:
Try putting csv in folder with PIDtoolbox.exe.

kmitchel, 2019-02-18:
Seems to have helped.  TY.

closed 2019-02-18 by kmitchel

kmitchel, 2019-03-24:
I usually use a version of BBE built off of the latest master branch commits...this seems to contribute to my loading problems.

bw1129, 2019-03-24:
The next version you can open bbl/bfl files directly. Also working on path related issues so this should all work better.