bw1129/PIDtoolbox #5: The readme did not mention PIDtoolbox also requires Curve Fitting Toolbox

DusKing1, 2019-02-11:
For the first time I run PIDtoolbox.m in matlab, it back a error says I need Curve Fitting Toolbox. 
I think you'd better mention that. 😃

bw1129, 2019-02-11:
Thanks. I’ll make note of that. You could run standalone version for now. Also make sure you’re running latest version (currently 0.12). I will figure out what it needs there and try to make a workaround.

DusKing1, 2019-02-11:
That's not a problem actually. I can restart my setup program and install Curve Fitting Toolbox.

bw1129, 2019-02-11:
I've updated the readme in response to this. Required toolboxes: control toolbox, signal processing toolbox, image processing toolbox and statistics toolbox.

DusKing1, 2019-02-11:
Much appreciate! 😃

closed 2019-02-11 by DusKing1