bw1129/PIDtoolbox #6: non-issue: request for Linux support

meoso, 2019-04-12:
i see Linux MATLAB run-time is available.  it would be a nice feature to have this project run native in Linux too.

bw1129, 2019-04-12:
thanks meoso. the thing is I'd need a linux machine and a version of matlab running on it to compile a linus version (not just runtime). That said, I should be able to install linus on a Mac using parallels or something right? I do have access to Matlab for linus from the University, so I will try to get this! cheers!

ghostface, 2019-04-26:
Yes I did some reading and you can't crosscompile you need to compile it from a linux matlab installation.
I am currently trying to setup matlab on my box using the trial version - would be awesome if we can get linux binaries tho.
Not sure if I can compile something with the trial, if I do I'll add the bins here.

ghostface, 2019-04-26:
I got it running under linux at least.
Caveats:

- blackbox_decode.exe is not found, I changed it in PTimport.m to blackbox_decode
- blackbox log files need to be in the same folder as PIDtoolbox - they won't load from anywhere else
-  scaling was messed up for me tiny matlab window, and huge PIDtoolbox fonts (i am running a 4k display with i3 as window manager so might not be an issue on other wms. I ran this to fix it somewhat in the commandline:

`
s = settings;s.matlab.desktop.DisplayScaleFactor
s.matlab.desktop.DisplayScaleFactor.PersonalValue = 2
`

bw1129, 2019-04-26:
You’ll need bb decode for Linux. Just go to bb tools in GitHub and pick that up. Then you change system function to run bb decode in linux.

Wrt font and other sizes, go to lines in UI control scripts that state something about proportion screen size. It’s currently .065 or something. Changing this will have a global effect on changing button and all ui widgets. I think it even affects fonts.

bw1129, 2019-04-26:
You might also need some toolboxes but hopefully you have em. Anyway, great job! Love to see how it works

watageek, 2019-04-27:
For linux,
Can't you just use a distro on a live boot usb. (linux mint for example with rufus.exe)
16gb usb is easy to find

bw1129, 2019-04-27:
@watageek I think you still have to have matlab for Linux installed with all the toolboxes. I have matlab for Linux. Have only Mac and Windows machines. I know I can install Linux on a Mac partition somehow but have not done it before. If u can point me in the right direction I might be able to get this moving.

watageek, 2019-04-27:
Info is to be verified, on top of my head... There is a "folder" called persistence when you do a usb key with rufus.exe on a windows OS. Adjust to suit your needs.

ghostface, 2019-04-27:
> You’ll need bb decode for Linux. Just go to bb tools in GitHub and pick that up. Then you change system function to run bb decode in linux.
>
> Wrt font and other sizes, go to lines in UI control [...]

Yeah I got running just fine - thx for the tip about the proportion size - will check that out.

watageek, 2019-04-27:
I run linux mint on a laptop PC on win10.  Dual boot. Grub menu is set to boot on linux default.

One of my friend is providing free unlimited 24/7 support. I am blessed... 😂

RustyToad, 2019-05-18:
@bw1129 Here is a good guide to dual-booting Mac and Ubuntu. I used to have a mac and this method worked well for me for years.

I only run Ubuntu linux now, but am happy to help test any way I can. I'd love to get this running.

watageek, 2019-05-18:
Post the link to that guide. 😚
 @RustyToad

RustyToad, 2019-06-20:
@watageek sorry, just saw your reply. I don't know if I forgot the link, or if it was filtered out. Here it is again:

https://www.maketecheasier.com/install-dual-boot-ubuntu-mac/

If that doesn't work, it's the first hit on Google for me for "dual boot Mac Ubuntu". It's on maketecheasier, and uses REFInd to manage the boot up.

meoso, 2019-06-20:
@bw1129 ... hey man, i see people telling how to dual-boot etc -- what a PITA...  If you do decide to support linux, then just install some linux distro (Xubuntu for ease) as a VM. (i..e Parallels if you already own it, or VirtualBox is free.) This is arguably "better" because you won't be outside of your normal OS environment, with the caveat that it's slower to run the VM instance, but there are some benefits such as copy/past or even sharing a folder between host-OS and VM.

But I digress, linux support would be great, but certainly should not take your priority. I can just as easily run a Windblows VM too.

bw1129, 2019-06-20:
@meoso ya, I would do it using ubuntu or something. either way Id need to extend the matlab license and install for that. that's the main issue. I think I have it free from my uni so may do that, but for now Im busy with basic upgrades. the other thing though is every operating system using matlab requires slightly different programming, so I cant just copy past the code, but have to tailor it for Win, Mac, and Linux separately. that's tedious and prone to errors. anyway. will do my best

qratz, 2019-06-27:
> I got it running under linux at least.
> Caveats:
>
> * blackbox_decode.exe is not found, I changed it in PTimport.m to blackbox_decode
> * blackbox log files need to be in the same folder as PIDtoolbox - [...]

So, could you compile and offer linux binaries that we can use with the linux version of matlab runtime?

ghostface, 2019-06-27:
> So, could you compile and offer linux binaries that we can use with the linux version of matlab runtime?

As far as I understood you need a proper license to do this sadly. The evaluation one doesn't allow you to pre-combile binaries.

bw1129, 2019-06-27:
@ghostface yes, I can get Matlab for linux. Working on it

qratz, 2019-07-05:
in the meantime, i am working on porting to octave... so far so good:
[image]

bw1129, 2019-07-05:
@qratz Dude that's awesome! I have a new version ready for windows. Would you like the source code to bring this up to speed on the latest? Great work!

qratz, 2019-07-05:
So far I have gotten the main view and the pid error view to work. Next I will try the spectograph. Once I have that working I'll put it on GitHub and try to keep track with the newest Version. The work is kind of a PITA, because there are so many tiny little things that work in octave but not in matlab, and vice versa. If you want to make life easier for me you could exclusively use the "set" and "get" methods on any GUI object properties, such as "foreground" and "fontsize".  As far as I know matlab accepts either something like guihandles.checkbox0.Value or get(guihandles.checkbox0, 'value') but octave only accepts the latter. Same goes for assignments via "set" vs. "=". That is like 90% of the changes I had to apply 😀.

bw1129, 2019-07-05:
@qratz I think the main reason I aborted octave was that it tended to be very slow plotting, but that might be resolved with newer versions. can you confirm whether it is faster? e.g. when clicking different traces it would think for 1-2 secs. I think it was an opengl issue with octave
https://savannah.gnu.org/bugs/?53845

qratz, 2019-07-05:
I will have to check that. For me, being a Linux only user, it was about having the toolbox at all rather than concerns about speed 😀. So far, I have only ran it on my laptop , which is an older i7 with Intel graphics, so basically no hardware accelleration. However, the startup time and initial loading of the BBL, feels very fast (have to go by feel here, no benchmarks available yet). But you are right, when I click on one of the plots on the main view it takes a second or two to display that plot zoomed in.

ghostface, 2019-09-08:
@qratz can you please share your progress on the octave port for linux?

qratz, 2019-10-01:
Hi everyone,
i had to suspend my work on the octave port for a few weeks, because i had to move :). i have been working my way through the menu system of pidtoolbox, starting with the main page and then the piderror and step response, leaving the last and biggest chunk last: the spectograph. There is a bunch of stuff there that is unsupported by octave. One problem i have not really focused on is the way the font sizes are calculated. This seams to be overly complicated in the code and leads to font sizes adjusting whenever i reopen a tool window. This looks extremly bad in octave, but the same problem is there in the windows version of original pidtoolbox, but not to the same extent as in octave.

Brian, have you noticed that too?

bw1129, 2019-10-01:
Yes, the font thing is unnecessary, but I tried to make it so that each time the screen size is changed, the fonts would change accordingly. This is especially useful for differences in visibility across different screen sizes, but it’s certainly not essential. My advice would be to fix a font size to something like 12 or 14 point an go with that.

bw1129, 2019-10-15:
@tiagolramos  I wish! thanks for the offer. I just need to install Linux and Matlab for Linux with the appropriate toolboxes, that's all. aiming to do that soon!

bw1129, 2019-10-17:
Would anyone here be willing to compile a version of blackbox_decode for Linux (Ubuntu)? This would really help. I’m hoping to compile Ptb in Linux very soon but would need the package blackbox_decode with it.

tiagolramos, 2019-10-17:
compiled in ubuntu 18.04

[blackbox-tools-linux.tar.gz](https://github.com/bw1129/PIDtoolbox/files/3737155/blackbox-tools-linux.tar.gz)

bw1129, 2019-10-17:
Oh man, thanks so much! Will try to compile Linux version of Ptb today

bw1129, 2019-10-17:
A BETA version of PIDtoolbox for Linux is available for download. Please follow the download instructions and let me know if it works. I was unable to test the executable directly (only tested from Matlab itself and it ran). 
https://github.com/bw1129/PIDtoolbox/releases/tag/v0.37

meoso, 2019-10-17:
installed with your included installer.
had to `chmod +x` a couple things
installed with `sudo` to default folders (`/usr/PIDtoolbox` & `/usr/local/MATLAB/MATLAB_Runtime`)
tried `./run_PIDtoolbox.sh /usr/local/MATLAB/MATLAB_Runtime/v97` from both the extracted `./main` & `/usr/PIDtoolbox/application`
result:
```
./run_PIDtoolbox.sh /usr/local/MATLAB/MATLAB_Runtime/v97
------------------------------------------
Setting up environment variables
---
LD_LIBRARY_PATH is .:/usr/local/MATLAB/MATLAB_Runtime/v97/runtime/glnxa64:/usr/local/MATLAB/MATLAB_Runtime/v97/bin/glnxa64:/usr/local/MATLAB/MATLAB_Runtime/v97/sys/os/glnxa64:/usr/local/MATLAB/MATLAB_Runtime/v97/sys/opengl/lib/glnxa64
Error using matlabpath
Modifying the search path is not supported by MATLAB Compiler. Remove functions that modify the search path from your MATLAB code. To make files visible to your deployed application, add the parent folder to your MATLAB session.

Error in path (line 109)

Error in addpath (line 86)

Error in PIDtoolbox (line 14)

MATLAB:mpath:PathAlterationNotSupported

```

tiagolramos, 2019-10-18:
i can try make a deb package after the installation script is fine tuned

vaughnd, 2019-10-25:
Same trouble with the 'modifying the search path' for me that meoso has.

mituritsyn, 2020-04-05:
@qratz hello, great work with octave port! Maybe will be better if someone could help you? Is your progress available on git or anywhere?
@bw1129 thanks for this progect. Is there any idea how to make linux reliases regular?

qratz, 2020-04-05:
> @qratz hello, great work with octave port! Maybe will be better if someone could help you? Is your progress available on git or anywhere?
> @bw1129 thanks for this progect. Is there any idea how to make linux reliases regular?

Since folks started publishing Linux releases I stopped working on the octave port. But that was not the only reason. I kind of got a brick wall with some of the analytical tools used in Matlab which are not available in octave, especially in the spectograph. Due to work I did not have the time to dig deeper into octave. But since we are now all under stay at home order I might start to work on it again. The thing that cost so much time, was actually the thousands of tiny octave-matlab thing. It would be a HUGE help if the original code constantly used set and get instead of "." To access properties.

bw1129, 2020-04-05:
@mituritsyn @qratz I think the Linux release had some issues and I couldnt find the time to resolve them. wrt octave, I actually had a running version, but what held me back was mostly the slow plotting. I may dig this up again and see if I can release what I have. I had the main stuff working (spectrograms and step plots). there were just aspects of it I wasnt happy with at the time. anyway, I'll dig into what I have and see if I can upload a decently working version. thanks for reminding me.

mituritsyn, 2020-04-06:
@qratz @bw1129 as I understood, you guys both make a huge work on octave independently. Just imagine, what could happen in coop or even with community help. Get out of the sandbox)

bw1129, 2020-04-06:
@MITuritsyn it’s not an issue with sharing. I simply didn’t want to release a bunch of messy code and this code is based on a very old version of Ptb so that would not be helpful either. I released Ptb so I have no issue with releasing an octave version. Just waiting till it’s functional enough to be useable, and time is precious these days.

flexd, 2020-06-09:
What is missing to make this work? Is there anything I can help with?

ghostface, 2020-09-20:
The problem is this:

```
executableDir = pwd;
addpath(executableDir)
```

It causes the compiled binary to start with the error

```
Error using matlabpath
Modifying the search path is not supported by MATLAB Compiler. Remove functions that modify the search path from your MATLAB code. To make files visible to your deployed application, add the parent folder to your MATLAB session.
```

ghostface, 2020-09-21:
Ok I did some reading but I am unable to test this as I can't compile without a license.

Changing the addpath to 
```
if ~isdeployed
    addpath(executableDir)
end
```
Should avoid the error.
I think matlab automatically includes the current directory in the searchpath so we might be able to live without this call anyway.

Can you please do a recompile with this change for testing please @bw1129 ?

bw1129, 2020-09-21:
Unfortunately there’s bigger issues than that. Linux is a no go at the moment.

ghostface, 2020-09-21:
It was working fine from what I could see when I was testing the non-packaged version with the evaluation license.

Is there a list or any more info on what those issues are? 
I am happy to try to help

bw1129, 2020-09-21:
I compiled it from a version of Matlab for Ubuntu on virtual machine. The problem I had was I could not test the app there directly so I was never sure what it was doing. I’ve no experience in linux so I’m really no help. The code is there. All I can say is grab it and try to compile yourself. Sorry

meoso, 2020-10-30:
progress :)
latest version `PIDtoolbox_v0.392` runs on Debian :)
but also i get this
[image]
which is likely a separate/external issue.

bw1129, 2020-10-30:
That’s an odd error to get because it speaks to an issue with one of Matlab’s built in functions, but only as it pertains to your operating system. It’s the first time I’ve seen it and Ptb is currently working on multiple versions of win, Mac, And Linux

JoeriCapens, 2021-06-10:
I've been using 0.392 on my Debian machine for a while and the new 0.4x version seems awesome but there's no Linux release this time? I'm also confused why there are no recently changed files in the Git source repository?

bw1129, 2021-06-10:
Unfortunately support for linux is just too much work for me for an open source project and I no longer have a working version of Matlab in linux. 0.392 is still there for Linux.

bw1129, 2021-08-03:
I managed to get a draft of upcoming PTB v0.44 compiled for Linux. Should be ready for release soon.

mituritsyn, 2022-01-22:
@bw1129 will be there linux binaries in latest releases?