bw1129/PIDtoolbox #7: Feature: Open bbl files from everywhere

jasc76, 2019-04-17:
Love the tool
but I got my bbl files organized, so being able to open bbl files from everywhere would be great.

bw1129, 2019-04-17:
It’s should work if you just copy/paste blackbox_decode.exe in with your log files.

jasc76, 2019-04-18:
Well, I got my Logs organized like this:
Copter / BF Version / Logs / date / .bbl
so having the exe in all these folders is possible but not convinient.

bw1129, 2019-04-18:
You must know that’s not an issue with pidtoolbox but it’s an issue with blackbox_decode. Your organization scheme will not work for plasmatree either. If you consider just putting all logs in same folder, Ptb then creates unique folder for all plots automatically.

jasc76, 2019-04-18:
oh... understand -> closed
hm, then it would be killer, if BF Config would have an auto rename function on download of logs so maybe the craft name and download Date/time gets added to easier handling. I hope I could program software :-( I always failed trying to code something....

bw1129, 2019-04-18:
Ya, those little things are sometimes really hard to program. There’s roadblack to certain things that make it almost impossible as a programmer. I did prefer earlier BF logs that all had unique names. Not they use a counter which I think always starts at 0 with an empty card. Even RC split does this. GP increments file number but has to retain the last in memory using and internal battery or something. Anyway, tricky

jasc76, 2019-04-18:
Of course putting logic on the FC would be tricky, but I think it can be done directly within BF Configurator. First there could be a BF config option to set default download folder (optional)
If the download is triggered via bf config (maybe special button) it could work like this:
- As you click on "Download Logs" BF connects via MSP
- reads craft name
- disconnects,
- goes to File mode
- gets the files
- renames maybe BFLOG_CRAFT_DATETIME_Index
- moves to Log Folder

bw1129, 2019-04-18:
no, I never said putting logic in the FC is tricky, I meant programming things like that can be tricky in general. WRT betaflight config, that has nothing to do with PIDtoolbox. Im confused. PIDtoolbox does not use betaflight configurator, but if you have suggestions for how the configurator might be made better, best to post this on their site.

closed 2019-07-04 by bw1129