bw1129/PIDtoolbox #8: PID Toolbox freezes on BBL file

0crap, 2019-04-17:
With some files the PID Toolbox freezes when you click on "load + run".

I attached one BBL file for which this happens.
The BBL file can be opened perfectly fine in BB Log viewer or Plasmatree tool.
So it seems the BBL file is not corrupt.

[PID Toolbox freezes.zip](https://github.com/bw1129/PIDtoolbox/files/3090838/PID.Toolbox.freezes.zip)

bw1129, 2019-04-17:
your log file is partly corrupted and is missing header info. It will load into explorer, but you can see the header info missing, which PIDtoolbox requires to properly run. 
[image]

bw1129, 2019-04-17:
I've seen this happen once before on a Radix FC. Not sure why it's happening for you. What log rate?

0crap, 2019-04-17:
[MATEKF405-STD](http://www.mateksys.com/?portfolio=f405-std), log rate nothing special 2K. (8K/8K looptime.)
Doesn't happen all the time, but it is really annoying when it's just that flightlog you want to compare. ;-)

0crap, 2019-04-17:
Anything you can do?
Or should I close this issue and open one on the Betaflight Git?

bw1129, 2019-04-17:
no it's a flight controller logging issue. I could try to make it run without header info, but header infor is needed to compute set point because it needs to know your rates, super rates and expo setting to do this, and also needs this to compute pid error. you even see the issue in BB explorer in the screen shot I added from your log. I have the matek405 and have not seen this issue, but it's certainly and issue with logging. Maybe post to BF team as an issue. Maybe they have more info for you

0crap, 2019-04-17:
OK, fair enough. Thx for your time looking at it.

closed 2019-04-17 by 0crap

guillermo3r, 2019-05-21:
I have the same issue and I think that my log is not corrupted. What else could be wrong?
[LOG00001.zip](https://github.com/bw1129/PIDtoolbox/files/3203877/LOG00001.zip)

bw1129, 2019-05-21:
@guillermo3r you can confirm whether log file is corrupted by opening in BBE and look at set up tap. if stuff is missing, then something went wrong during recording. see if it's better with a different Sd card, and dont log too high. 2k is plently

Jdjacob3, 2022-09-10:
@guillermo3r @bw1129 I was just working through a very similar issue with my Matek F405SE FC. I had gone in with my computer and deleted a log file from the sd card. The logs after that didn't want to open in blackbox explorer OR PIDtoolbox. I removed the sd card, formatted it, reinserted it in my FC, did a flight, and I could successfully open that in PTB. 

If you have an SD card, format it and try again. Don't delete any individual Log files

SolkGit, 2023-02-25:
Had the same issue with MatekF405STD on my Darwin 129 - PTB could no load BFL files. Formatting SD card solved the issue. Now works fine.