bw1129/PIDtoolbox #9: Does not run. Only splash image shown

Cleric-K, 2019-04-18:
Hi,
I've just installed manually the `9.3` matlab runtime.
Running `PIDToolbox.exe` displays the splash image and then the process silently ends.
I don't see any logging in the Windows logs. Is there a way to run in console mode with logging?

Cleric-K, 2019-04-18:
I see that the matlab compiler can generate console version of the `.exe`. Can we do that? That should give some clue what's happening.

bw1129, 2019-04-18:
Try using the provided installer rather than downloading manually.

Cleric-K, 2019-04-18:
I tried that but the behavior is the same. So all cases are:
* manually installed runtime and running directly from `main` directory.
* manually installed runtime and using your installer. It detects that the runtime is already installed and uses it.
* no pre-installed runtime and using your installer. It downloads and installs the runtime.

In all cases the behavior is the same: The splash image appears and then the process ends.

We need to get some logging.

bw1129, 2019-04-18:
it can be stubborn. I dont understand the intricacies of runtime, but I think you have to remove/uninstall the previous, then for good measure, restart your computer, then do a complete new reinstall from the provided installer. I just put the entire folder on the desktop, and install from there, but I dont think it matters. try that and let me know how it goes.

Cleric-K, 2019-04-18:
I tried now, but it's the same.

Is it possible to generate console version of the `exe`?
Do you use the `mcc` compiler? It should be possible to generate console version with `-m` argument instead of `-e` (which generates Windows native app).
https://www.mathworks.com/help/compiler/mcc.html

bw1129, 2019-04-18:
that's really unfortunate man. 
no I dont know how to generate a console version. I use the build-in app compiler gui. there are no options you're mentioning. I can look into it.

Cleric-K, 2019-04-18:
I'm trying now at my office PC (Desktop Win10 64).
Will try later at other PCs - I assume it will work there, so it's no that of a problem, but it's just curious what could be the reason. Should be something quite specific to the current configuration (although it's pretty standard PC).

bw1129, 2019-04-18:
I think windows is holding something from the manual install that's not letting you past. there are now many people using it with no issues so that makes it a tricky thing to solve. Im sure it has something to do with paths, but Im not an expert in dealing with this. and the real trick is I can never know how its going to work on everyone's computer. I hope it works on your other windows machine

Cleric-K, 2019-04-18:
Can you tell me what version of MATLAB are you using? Is this capability of producing compiled version inbuilt or a separate product?
I'll see if I can get a MATLAB installation and compile myself.

bw1129, 2019-04-18:
Im on 2017b, but you'll need a bunch of toolboxes to run it. you need those toolboxes to compile too I think. Well, you'll for sure need compile-related toolboxes. But if you have matlab, then you can just run pidtoolbox from there. It's way faster if you just run from matlab command prompt.

bw1129, 2019-04-18:
let me know if it worked on your other machine??

Cleric-K, 2019-04-19:
I tried it on a VirtualBox on the same machine that has the problem and it works there.
It's clear that there's something with this setup that breaks it, but it's not possible to tell without some kind of logging.

When possible I'll try compile myself the exe.

watageek, 2019-04-30:
I had issue on two computer. Splash screen close then nothing appear. 

I5 650
Win7

AMD c50
Win10

On both I disabled real time antivirus, and I started flawlessly.

Both are running MSessential antivirus/window antivirus

Cleric-K, 2019-04-30:
Interesting. I'll try that, but probably next week I'll have access to the problematic pc.

ReefsnRidges, 2019-06-29:
Having the very same problem with windows 10.  Tried disabling antivirus, uninstall and re-install, reboot and all I get is the splash screen?  I also did  install the runtime file.

watageek, 2019-06-29:
The most possible issue is more than one instance of antivirus running or more than one program.

If you see the splash screen it is most likely you need disable real time protection.

If it is running on a 9watt cpu, anything can run it.

Open the task manager and see if find the culprit

ReefsnRidges, 2019-06-29:
Thank you for the suggestions, very much appreciated as I so badly want to use this tool however, same result.  From the task manager I could not see anything active when trying to launch toolbox.  Cleared the history and tried too, nothing except the error msg ".  At a loss...

bw1129, 2019-06-29:
@ReefsnRidges
Did you use the installation file to install the entire program? Do you get an error message or just hanging on splash screen on startup?

ReefsnRidges, 2019-07-02:
Yes, used the install file for everything.  Two days ago it decided to work.  Today again, same story.  No error message other than a chime from windows 10 when it try's to open and if I watch the task manager, it only pops up briefly when the splash screen is closing and indicates a "program not responding".

ReefsnRidges, 2019-07-03:
Brian, don't want to jinx myself here but wanted you to know I have it working.  Cleaned my registry thinking maybe left over files from to many install/un-installs and it worked immediately.  Thank you for all you do including trying to get me going here, much appreciated!  You guys are so giving of your time and expertise and I feel like you don't get near enough credit for what you do.  THANK YOU!!!

Cleric-K, 2019-07-03:
@ReefsnRidges Can you specify what registry entries you have deleted? I want to try it with a machine that has the same problem.

bw1129, 2019-07-03:
@ReefsnRidges Thanks very much man! greatly appreciated. I'm glad to figured out the issue. I will keep this thread open so others can see a potential solution if they have a similar issue.

ReefsnRidges, 2019-07-03:
The problem started when I tried to move the files (Pidtoolbox) into another directory as it worked prior to this move.  I then did countless install/unistall trying to get this to work, which further exasperated the problem.  I can't tell you specifically what files.  I can tell you what tipped me off was when doing an uninstall, I got a message that all related files were not uninstalled from the registry, and that I'd have to do that manually.  I have the free version of "CCleaner" so ran the registry cleaner which deletes files not associated with anything (after making a back-up of those files of course via the program).  After a scan I noticed in the files to be deleted there were several files from "Pidtoolbox" and once deleted, the Pidtoolbox came to life. Hoping this does the same for you as it is a great tool to have.  Best of luck!

ReefsnRidges, 2019-07-04:
Update: Not working again today.  Thanks everyone for your help.  Just to unreliable and time consuming at this point for me to continue, the white flags going up.

bw1129, 2019-07-04:
@ReefsnRidges is it always an issue of the program not starting, or are you mixing in issues with hanging while loading log file? Also remind me is this Mac or Windows?

ReefsnRidges, 2019-07-04:
Program not starting on Windows 10 PC. Splash screen pops up, Windows chime sounds and splash disappears. Within task manager it pops up as PIDtoolbox and a message “program not responding.

lylus, 2019-09-20:
win10 64bit 遇到了同样的问题。望能解决。

UglyBabyGaming, 2022-02-16:
Hi I had this working no problem at the weekend.. few days later just shows splash and shuts off
tried re-install no joy

bw1129, 2022-02-16:
I think windows updates are blocking it. Go to folder where the executable is and make sure it had full read write access

UglyBabyGaming, 2022-02-16:
No joy even run as administrator (did full premisionin the propertys tab under security)

bw1129, 2022-02-16:
I suspect latest windows isn’t playing well will 2017 Matlab runtime. I may begin compiling with latest Matlab.

UglyBabyGaming, 2022-02-16:
Using Windows 10 it happened before on an older version last year as well.
Hope you get it fixed its easy to use when working thanks.

bw1129, 2022-02-16:
Unfortunately it’s not me. There’s nothing in PTB code related to that. It’s a ML runtime windows issue. Some fix it by reinstalling

Message ID: ***@***.***>

Guidus93, 2022-02-21:
I have the same issue.
I reinstalled the runtime, try to download from Matlab website also and can't fix. I only get the splash screen. 
I also cleaned the register. 
I have Matlab 2021b installed and works if i execute the script from matlab direclty.

UglyBabyGaming, 2022-02-21:
> If you have Matlab you can run it from the command window. But you will need special toolboxes. I’m guessing you mean you have Matlab runtime 2021b. Runtime is not exactly Matlab. In any case you can’t run the program on a version of runtime that’s different from the version of Matlab in which it was compiled. I will be compiling it in the latest version of Matlab for windows soon. I suspect this is the problem since it’s compiled on a 2017 version.

I will try it out I was asking on the TBS lounge someone said it stopped and just started working again later for no reason??? This tool is so useful for tuning Ya Genus just all need the reliability of it working looking forward to the updated version.. ps basement tune soooo locked in on test flight. can you do a vid on tuning Dmax Bf 4.3? as only tuned Dmin and left Dmax off… the tune is so good I actually dropped my rates by 180deg/sec

closed 2023-05-11 by bw1129