bw1129/PIDtoolbox #14: Cannot load log recorded on desk (not real flight data)

voroshkov, 2019-05-24:
I've recorded a specific log by setting "blackbox_mode = motor_test" and spinning a single motor from BF Configurator Motors tab. This log refuses to load in PID toolbox for some reason. 
CSV gets generated and contains a lot of zeros, which may be the reason.
[btfl_001_gyro_scaled_more_noisy.01.zip](https://github.com/bw1129/PIDtoolbox/files/3215895/btfl_001_gyro_scaled_more_noisy.01.zip)

bw1129, 2019-05-24:
Check to see if the log file is corrupt. Open in BBE and look to see if setup info exists. If missing anything, Ptb can’t compute critical variables like pid error

voroshkov, 2019-05-24:
yes, it's working fine in BBE. Here's the BBE log itself: 
[btfl_001_gyro_scaled_more_noisy.zip](https://github.com/bw1129/PIDtoolbox/files/3217604/btfl_001_gyro_scaled_more_noisy.zip)

voroshkov, 2019-05-24:
It just hangs upon loading .css

bw1129, 2019-05-24:
ok, so I ran it directly in matlab. it's missing a string that is usually there. I used a string that should always appear as a marker for the end of the header list of variables, but for whatever reason it doesnt exist in your file. Does this issue happen on all your log files or only this one?

bw1129, 2019-05-24:
it's an easy fix in matlab. Im happy to compile and send you revised version. but you have other odd things as well. there's no motor sigs in the file, it's sampling at 6Khz, and % throttle is all messed up. what kind of FC is it? This all seems odd to me

voroshkov, 2019-05-24:
I bet that odd log format might be the limitation of the "MOTOR_TEST" logging mode. All logs I take on the same FC in "NORMAL" mode are processed perfectly in PIDtoolbox. Thanks for your work!

voroshkov, 2019-05-24:
So, Brian, would it be possible to make the PIDToolbox recognize and load both these types of logs?

bw1129, 2019-05-24:
yes it's possible. just not able to get at it for a while. very busy at work

bw1129, 2019-07-04:
I am closing this issue because it is resolved and will be fixed in the next version of pidtoolbox (v0.30).

closed 2019-07-04 by bw1129