bw1129/PIDtoolbox #15: Feature request: time-based spectrogram

voroshkov, 2019-05-24:
Would be quite useful to have a possibility of choosing "%time" as X-axis in the spectrogram. 
I.e. having distribution of frequencies in a signal by small timeframes. This would help me a lot in analyzing the frequencies of gyro noise in my setup.
Thanks a lot for the useful tool!

bw1129, 2019-05-24:
Would %time be useful or actual time so that you could trace back in the gyro the points where some noise occurred? I’m assuming the latter because %time would require some guessing. Also time would have to be binned since you need the data segment to be a certain length to compute fft. Might work though. I do see the value in having a plot like this. That said, the spectrograms involved some smoothing across %throttle, but that wouldn’t make sense if x axis was time, or else you’d be smoothing out important differences at specific times. Not sure how this would look. Do you understand what I’m thinking? Thoughts?

voroshkov, 2019-05-24:
Sorry, I misinterpreted the % sign in front of "throttle" :). I meant absolute time scale in seconds on X axis. Not sure what is meant by "binned" time, I'm not so strong in math, sorry. I guess it might be a fixed "window" of, say, 256 samples - 128 at each side of the specified time. Maybe these windows may overlap for neighbor time counts, I'd try and see how the result looks like. 
As for smoothing - not sure if it is needed. I'd also use "trial and error" method :)

metametaclass, 2019-05-24:
Something like that:
[image]

Above spectrogram is generated by matplotlib from gyro data with code like this: 

`plt.title("yaw_spectrogram")
powerSpectrumYaw, freqenciesFound_yaw, time, imageAxis = plt.specgram(gyro_yaw, Fs=sfreq)
plt.colorbar(imageAxis)`

Default size of fft window for specgram is 256 with no overlap and hanning window is used by default.

bw1129, 2019-05-24:
@voroshkov so basically to do fft you need a minimum of say 300-400ms time segments of data. any smaller and noise below say 5Hz or so would be meaningless. that should be fine though. It's certainly doable and I can see the usefulness. 

@metametaclass interesting. looks like some artifacts running through the plot though, maybe due to lack of tapering of segments ??

metametaclass, 2019-05-25:
> @metametaclass interesting. looks like some artifacts running through the plot though, maybe due to lack of tapering of segments ??

I`m not sure what causes these side lobes. They are at a constant distance (~110 Hz) from main frequency component on all spectrograms (roll/pitch/yaw) and they don`t change with different FFT lengths or window selections.

v-like shapes at middle throttle values - aliasing of some high frequency noise from motors (more then 4 khz Nyquist frequency for 8 khz gyro).

bw1129, 2019-05-25:
If this is derived from plasmatree, there is subsampling which repeats data segments slightly shifted in time as it progresses through the data. Couple that with no hanning taper, and you could get such artifacts. A taper is critical for short segments of data because the ends of each segment may not fall to zero and this will produce an artifact during fft. If it’s not a taper issue, it’s certainly something that is carried over from subsampling (repeated sampling of overlapping data segments), which is necessary for the other plotting techniques (% throttle) but may not be ideal in this case

metametaclass, 2019-05-25:
With simulated frequency sweep signal there are no obvious artifacts: 
[image]

By default specgram is using hanning window so I think that artifacts are from gyro or flight controller preprocessing stages.

Sound spectrogram from microphone mounted on bench shows mainly RPM harmonics and no equal spacing artifacts either: 
[image]

bw1129, 2019-05-25:
Looks great! I’ll see what I can do with Ptb.

closed 2023-05-11 by bw1129