bw1129/PIDtoolbox #16: Some logs open, others hang on "Converting csv numerical data for LOGXXX.BFL"

SlipstreamFPV, 2019-05-27:
Some of my logs don't open. 

It appears that occasionally my logs are missing a few bits of data from the header and this might contribute to them not working at all.

After selecting the logs that don't work then hitting "load+run" the pop up windows hangs on "Converting csv numerical data for LOGXXX.BFL" 

The windows error chime sound plays, but no error box pops up.

I have attached a zip with one working log and one that doesn't work for me.

Thanks for all of your hard work @bw1129 It's a great tool.

[Logs.zip](https://github.com/bw1129/PIDtoolbox/files/3222478/Logs.zip)

bw1129, 2019-05-27:
yes, several people mentioned this. It's usually as you said, the header info is missing. it's too bad. I could make Ptb force through it anyway, but then many things would be missing, and it would require quite a bit more programming to have it skip over missing data when plotting certain things. thanks for your feedback. I will work on this

bw1129, 2019-05-27:
[image]
I think these broken snippets mean your SD card failed. this might explain why you're also missing header info.

SlipstreamFPV, 2019-05-27:
Thanks for your response. Just trying to use cheaper (but still certified) as cards but maybe they’re not quite fast enough. I’ve lost too many good blackbox cards when they pop out the side!

atomiclama, 2019-06-04:
I'm seeing this problem as well and thought it might be my PC. Would it be possible to just abort the loading/converting with an error message?

bw1129, 2019-06-04:
@atomiclama apparently it is an issue with Betaflight 4.0 that is being resolved.

atomiclama, 2019-06-06:
@bw1129 That's good to know thanks, but would it be possible for your tool to gracefully handle corrupt log files and abort, rather than just hanging?

ecob81, 2019-06-28:
Hi, does anyone know if this has been fixed yet or maybe have a link to the issue in betaflight (I searched but can't find anything). Took 4 logs in BF 4.0.2 today and they all hang like this.

bw1129, 2019-06-28:
@ecob81 best to message betaflight. I think it is fixed or they are working on it.

bw1129, 2019-07-04:
I am going to close this issue as it seems to be essentially a BF logging issue and should be dealt with there.

closed 2019-07-04 by bw1129