bw1129/PIDtoolbox #17: Mac Os X install

orlovskiy, 2019-06-04:
So I got some errors during install which is:
[image]
And another one kinda like this. After those two I got "Installation successfull" message. After finished installation there is an empty folder PIDToolbox in Applications. When I run the program from Downloads folder(or copying contents of "main" folder to created empty folder in Applications) it is stocking on 
[image]
Also I have installed MATLAB Runtine R2018b downloaded from MATLAB's site. 
Completely stocked, but very interested in trying this.

exocode, 2019-06-19:
Only wanna join, have the same issue:

```
Source '/private/var/folders/k9/kt8cnmqj5h10wvh142d8znv40000gn/T/AppTranslocation/8D3E9C1A-55FB-41B8-BF8D-816508C109B8/d/MyAppInstaller_web.app/Contents/Resources/bundle.zip' does not exist
```

When installation wsa finished I saw this:

```

Product Configuration Notes:  In the following directions, replace MR/v95 by the directory on the target machine where MATLAB is installed, or MR by the directory where the MATLAB Runtime is installed.  If the environment variable DYLD_LIBRARY_PATH is undefined, set it to the following string:  MR/v95/runtime/maci64:MR/v95/sys/os/maci64:MR/v95/bin/maci64  If it is defined, set it to the following:  ${DYLD_LIBRARY_PATH}:MR/v95/runtime/maci64:MR/v95/sys/os/maci64:MR/v95/bin/maci64 
```

exocode, 2019-06-19:
If I load any file on my MAC I get this after pressing "load + run". Why is it calling for an .exe file? Mac users cant open exe files without installing additional programs.

[image]

bw1129, 2019-06-19:
@exocode it's a generic error label, but it's fixed in later version. Ignoring the .exe part, the message is still valid. it does not see blackbox_decode in same folder as log files you selected. try placing the log files in same folder as those program files (the mac versions of bb decode and pidtoolbox) then select the log file from that folder

krzysztofmatula, 2020-02-22:
Hi,
How do you pass over the "File '/private/var/folders/4t/9071clns5nlbxnqwzz04hyfc0000gn/T/AppTranslocation/6D0C468D-5FA9-455E-941F-F846822B65B7/d/MyAppInstaller_web.app/Contents/Resources/bundle.zip' does not existthe " message?

I'm trying the latest PIDtoolbox release and I'm getting exactly the same as in this issue description...

Is there some workaround?

krzysztofmatula, 2020-02-22:
Answering my own question: this helps:
```sudo xattr -d com.apple.quarantine MyAppInstaller_web.app```

closed 2023-05-11 by bw1129