bw1129/PIDtoolbox #18: PIDToolbox cant read BF 4.04

jasc76, 2019-07-07:
Hi

I tried to analyse a BF4.04 Log 
Name/Version: 4.0.4
Binär: betaflight_4.0.4_MAMBAF722.hex
Datum: 01-07-2019 14:36

The csv is created and PTB reads it and the progress bar disappears but then PTB just hangs
Tried a older Log on the same computer and works fine

Regards
Jan

bw1129, 2019-07-07:
Can u upload the logfile so I can take a look?

jasc76, 2019-07-07:
here it is
FC is mamba 722 mini
frame: Diatone Exorcist
sadly no Gyro scaled...forgot :-)
PIDs / Filters all BF 4.0.4 stock
Motors Flywoo 2207
[btfl_002.zip](https://github.com/bw1129/PIDtoolbox/files/3365592/btfl_002.zip)

bw1129, 2019-07-07:
I have no issue opening on v0.23. What version are you using? 

[image]

jasc76, 2019-07-07:
yes 0.23
 
Did you try session 2 ?

[image]
I hear the windows "bing" sound and progress bar shows 100% (see picture)

bw1129, 2019-07-07:
yes. Log 1 top, log 2 bottom. 

[image]

bw1129, 2019-07-07:
try grabbing the latest exe again from 
https://github.com/bw1129/PIDtoolbox/releases
You dont have to download runtime again, just grab PIDtoolbox.exe and replace yours with this one.

jasc76, 2019-07-07:
Did that, doesnt help. After selecting the session PTB works (cpu+ram) when Progress bar is at 100% I get the windows notification sound, cpu+ram drops to 0 but nothing else happens

jasc76, 2019-07-07:
Other logs work, also session 1

bw1129, 2019-07-07:
it's working on 0.23 on both my Windows and Mac machines. Not sure what to say, but there's nothing I can do on my end if all is aok here, and I've not heard other reports of any issues. 0.30 will be released in the coming weeks so maybe give that a try, and hopefully that works for ya. that's about all I can say at this point.

jasc76, 2019-07-08:
Thanks for your effort! 
Then, I guess, my Computer has an issue...

m013, 2019-07-17:
the same problem for me. stucks at converting (progress bar instantly goes 1/3 and nothing)
on both my mac and windows pcs

version 0.23

bw1129, 2019-07-17:
@m013
Can you upload the log file? I’ll take a look.

m013, 2019-07-17:
@bw1129  , sure.

[main.zip](https://github.com/bw1129/PIDtoolbox/files/3401586/main.zip)

bw1129, 2019-07-17:
@m013, ah ha. I've had this happen to me before. for some reason blackbox records a whole bunch of dud log files woth no information, amongst the few real log files that are near the end. the problem arises when blackbox_decode.exe (part of BB tools) only extracts up to the first ~30 logs and will not convert numbers higher than this. Im not sure quite why, but I have to ask about this potential issue to the developers of blackbox_decode.exe, and/or why blackbox records these junk files. This might be a new issue for betaflight. I will look into it. I should mention, I had the same problem on earlier versions of BF. I have a hunch that onboard logging devices add an empty log file to the onboard flash every time the copter is plugged in, and unless a log recording is selected, it just leaves the empty log on there. blackbox_decode.exe might be just programmed to ignore anymore than ~30 logfiles, presuming that this would not be possible memorywise. anyway, let me look into it. 
[image]

m013, 2019-07-17:
I have been getting this crap data since forever on fcs with onboard memory chips. Bb explorer and plasmatree pid analyzer recognize it correctly tho

ijustwannafly, 2019-08-26:
This is now happening to me on certain log files.

Bf 4.0.4.  Pidtoolbox v0.31.  

Blackbox explorer can open and read the log file fine but pidtoolbox just hangs with no progression.  Other files are no issues.

[2.zip](https://github.com/bw1129/PIDtoolbox/files/3539195/2.zip)

bw1129, 2019-08-26:
@ijustwannafly  check my comment above. open in BBE, look to see if more than 30 files in drop down. if so, tis is the issue. I am still working on it, but bb_decode sets limit on # of logs to extract in a file. BF create a new blank file each time FC is enabled. only solution for this issue right now is to ensure you logging flash drive is cleared before collecting new logs.

closed 2023-05-11 by bw1129