bw1129/PIDtoolbox #19: [FEATURE] clean file structure in app folder

exocode, 2019-07-12:
In order to keep log files in order, it would be nice not get forced to keep all files in the app's root folder. I don't know how this behaves on a PC, on a MacOS I do have to put all **logs in the root folder of the app**, which is going quite messy very early.

bw1129, 2019-08-08:
@exocode you should be able to put log files where ever you wish, as long as "blackbox_decode" is in the same folder as the log file.

ijustwannafly, 2019-08-10:
On the subject of directories, can the output folders NOT be created UNLESS we actually choose to save the graphs or other data?  Currently it creates a folder for every log file I open even when I do not save the graphs.

bw1129, 2019-08-10:
@ijustwannafly 
done in v0.30