bw1129/PIDtoolbox #21: Handling logs in different source directories

ijustwannafly, 2019-08-17:
Currently I have pidtoolbox working but with a different folder structure all around.

The "main" directory is nested under "root..pid_tuning\pidtoolbox\main"

My blackbox logs (for organization) is nested under "root...blackbox_logs\directories"

With this structure, I can load pidtoolbox and choose logs from within the blackbox_logs directory and run the process with no issues.

As long as two log files are within the same directory there are no issues.

However, if I have folder nested as "root...blackbox_logs\quad1\series1" and "root...blackbox_logs\quad1\series2" is where issues start.

If I open two logs from within the series1 directory pidtoolbox does not complain and runs both files.

If I open a log from series1 under A and a log from series2 under B then pidtoolbox complains that the log file needs to be under the "main" folder with blackbox decode.

bw1129, 2019-08-17:
I just tried that and had no issue here. Blackbox_decode.exe must be in both folders, or any folder where u intend to place and use bbl files.

ijustwannafly, 2019-08-17:
That's the confusing part.  

The way I have been using it so far with no issues is that the blackbox_decode.exe is only located in one place, within the main directory of pidtoolbox.

With the blackbox_decode.exe only in the main directory, I can choose two blackbox log files from any other directory that does not have the blackbox_decode.exe file and pidtoolbox will process the files.

However, if I choose two blackbox log files from two separate directories it fails.

It will work, if blackbox_decode.exe exists only in the main directory, the two log files are in another directory with no blackbox_decode.exe, AND the two log files are under the same directory.

If I have series1 and series2 directories, neither of which have blackbox_decode.exe, the files can be opened from these directories provided that both the A and B source originate from the same directory.

Basically it makes no sense to have to keep moving/copying files back and forth especially when you are trying to keep the logs organized for archive and progression.

Now , this is on a Win10 machine.  When I use my android tools for example, I only have one instance of adb and fastboot located anywhere on my machine and these exes are referenced under the PATH variable so that I never need to drop to its directory to call its commands.  Is that possible here?  I am thinking that may be a solution to this instead of replicating the blackbox_decode.exe file everywhere.

ijustwannafly, 2019-08-17:
This can be closed.  I managed to re-organize and link directories in a way that it is not an issue.

Thank you.

closed 2019-08-17 by ijustwannafly