bw1129/PIDtoolbox #25: non-issue: request for reading Openlog files

ResinSmoker, 2019-09-27:
Hello, I've recently installed an Openlog serial data recorder on an FC without it's own logging. While a flight does produce a log, it's is in .txt format and thus not usable within Toolbox / BFL Analyzer.

Would it be possible to add support to read .txt data logs?
[LOG00023.TXT](https://github.com/bw1129/PIDtoolbox/files/3661821/LOG00023.TXT)

Thank you in advance!

RS

ResinSmoker, 2019-09-27:
Edit... discovered a few things that solved the problem.

1) The data rate in the logger wasn't set to 250k
2) The file extension can be changed and then read normally.
3) The log has a significant number of gaps. Guessing that the Openlog can't keep up with a 1k rate.

bw1129, 2019-09-27:
Yes I think the breaks in logging will not work in Ptb. Log at 2k

SyRexFPV, 2019-11-01:
> Edit... discovered a few things that solved the problem.
> 
> 2. The file extension can be changed and then read normally.

- how do you change the file extension to .bbl ?
- what firmware do you have for your openlog ?

Thanks

bw1129, 2019-11-01:
This is new to me but yes just add the .bbl extension and you’re good to go apparently. Thx for the confirmation

SyRexFPV, 2019-11-01:
thank you for your reply. I think I have a problem with my openlog. I think I do not use the right firmware. My log.txt file has strange characters after a few lines. I can open it in blackbox explorer but the result is strange, as if there was a problem of scale.
In any case, I can change the file extension in .bbl but I can not read it in PTB, I have error messages loading.
I will see to flash again  my openlog.
If you know which firmware to use to log a fc kiss V1 ... I think I will try "openlogV3blackbox" ...
Thank you

bw1129, 2019-11-01:
Check to see if you can convert the logfile to a csv file using blackbox tools. If this fails then we know why it fails in Ptb and there’s nothing I can do about it. If it works, then the fix is on my end

SyRexFPV, 2019-11-01:
With BlackBox Explorer, I can convert the logfile to a csv file. 

[LOG00255.TXT.csv.zip](https://github.com/bw1129/PIDtoolbox/files/3797920/LOG00255.TXT.csv.zip)

SyRexFPV, 2019-11-01:
My log
[LOG00255.TXT](https://github.com/bw1129/PIDtoolbox/files/3797927/LOG00255.TXT)

bw1129, 2019-11-01:
turns out that logfile has messed up or limited header info. unfortunately Ptb cant work without that.

bw1129, 2020-01-01:
so the main fix here is to rename the 'txt' extension to '.bbl' and should be good to go. I will consider this issue resolved. closing.

closed 2020-01-01 by bw1129