bw1129/PIDtoolbox #27: MacOS Catalina issues

swampmonster, 2019-10-14:
It would seem that the matlab runtime you are compiling with is not compatible with the latest version of macos.
I tried downloading the latest version myself (2019b) and using that but that just results in errors like:
dyld: lazy symbol binding failed: Symbol not found: __Z11PopDialoguePKcS0_

Would it be possible to compile with a newer version or do you not have access to that?
2019b seems to be the only one that supports catalina.

bw1129, 2019-10-14:
That really sucks! That’s Mac for ya. They change operating systems more often than people change their socks, and with every iteration, they no longer support the same software. This would require me to get a new version of Matlab, because this is what dictates the runtime version required. Unfortunately if I do that then the current users will have to upgrade their runtime (which is a huge download), and there is no guarantee it will work for them on slightly older version. By older it’s 2018, so why Mac won’t support software that’s a mere 1 year old makes me wanna stop supporting Mac completely. The best option I can give you is to download the Linux version when available. This is the priority now.

swampmonster, 2019-10-14:
That's entirely fair on your part so no worries. I can just run it in a vm for the time being