bw1129/PIDtoolbox #28: Telegram bot integration

LuisDiazUgena, 2019-10-31:
Hi!

I'm developing a bot for plastmatree which uses that tool and telegram bot api to convert log files.  This is kind useful for pilots who doesn't own a laptop, don't want or can't bring a laptop to the field.

I was taking a look at your work to make another bot to integrate it, and I got some doubts. I see that the bot runs on matlab and draws matlab windows to show the output, right? 

I was wondering if it could be possible to output that same windows to an image directly, that way i can run it on my ubuntu server, and send the images back to the users. That would be awesome!

Thanks!

bw1129, 2019-10-31:
Ptb was compiled from Matlab, and requires specific, free, Matlab runtime, which contains the guts/functions of matlab, without the user ability to do anything with those functions. It simply allows for deployed Matlab programs into a format for different operating systems. If you have Ptb working on your machine, or better yet, if you have Matlab with all the required toolboxes to run Ptb from Matlab command window, then I can imagine it might be possible to set up some kind of a “bot” that allows users to access that from your computer in some way. This is out of my expertise but look the matlab source code is there so be my guest. I’d love to see it work. It would most certainly be awesome!

LuisDiazUgena, 2019-11-04:
Hi there,

I understand what you're saying and it was my intention to make it work with a bot. I'll digg into the matlab code, but I have a very limited matlab knowledge. Would be awesome if you can give some support with matlab code issues I know I'll face.

bw1129, 2019-11-04:
Well I have no knowledge of bots so that makes two of us. Can you explain to me how a bot works and I might be able to know whether it would be possible? Matlab is a high level proprietary language so I have my doubts the matlab developers would allow it to operate via some 3rd party software but please if you could describe to me what the bot would do exactly

LuisDiazUgena, 2019-11-04:
I know how telegram bots work, that's no problem at all. My issue is with Matlab. As I imagine the working Algorithm for this integration, the bot will recieve a Blackbox file, download it and then run PTB on the command line. PTB should export results to images and then, the images will be taken by the bot and sended back to the user who sended the message. 

It could be great if we can use something like Telegram for chatting about this, will be much efficient that git issues.

closed 2023-05-11 by bw1129