bw1129/PIDtoolbox #29: macOS Catalina Compatibility

c3r63ru5, 2019-11-03:
Hey,

just upgraded to Catalina and can't get PIDtoolbox to run.

The Matlab installer hangs (downloading) and the PIDtoolbox app won't execute without errors.

Thanks,
Sebastian

bw1129, 2019-11-04:
this has been noted. thx

c3r63ru5, 2019-11-04:
Reinstalled MATLAB 2018b. Got the following output when running the start script.

```
./run_PIDtoolbox.sh /Applications/MATLAB/MATLAB_Runtime/v95/
------------------------------------------
Setting up environment variables
---
DYLD_LIBRARY_PATH is .:/Applications/MATLAB/MATLAB_Runtime/v95//runtime/maci64:/Applications/MATLAB/MATLAB_Runtime/v95//bin/maci64:/Applications/MATLAB/MATLAB_Runtime/v95//sys/os/maci64
dyld: Symbol not found: _mecab_get_feature
  Referenced from: /System/Library/PrivateFrameworks/CoreNLP.framework/Versions/A/CoreNLP
  Expected in: /Applications/MATLAB/MATLAB_Runtime/v95//bin/maci64/libmecab.dylib
 in /System/Library/PrivateFrameworks/CoreNLP.framework/Versions/A/CoreNLP
./run_PIDtoolbox.sh: line 29: 42445 Abort trap: 6           "./PIDtoolbox.app/Contents/MacOS/PIDtoolbox"
```

bw1129, 2019-11-13:
FYI, I will soon be upgrading Matlab for Mac to the latest 2019 version, and will re-compile a new version of Ptb. This should make the "myinstaller" installation file use the most recent version of Runtime and associate Ptb with that version. I am hopeful that this will fix the compatibility issue with Catalina, so stay tuned. If it works, all Mac versions of Ptb will be based on this version of Runtime moving forward.

joshuahardee, 2019-12-08:
Hi @bw1129 have you been able to update Matlab for Mac to the latest 2019 version? I too am anticipating this release.

bw1129, 2019-12-22:
OSX Catalina users: Beta version of Catalina-friendly version of PTB available (with suffix "osxc"):
https://github.com/bw1129/PIDtoolbox/releases/tag/v0.39

This was compiled with the latest (2019) version of Matlab, which has been reported to be Catalina compatible. That said, I have not tried it in Catalina, but it works fine in earlier versions of osx, so I'd appreciate feedback as soon as you try it. NOTE: this requires a full install using the supplied "MyAppInstaller_web.app" under the "runtime_installation_file" folder.

c3r63ru5, 2019-12-23:
> OSX Catalina users: Beta version of Catalina-friendly version of PTB available (with suffix "osxc"):
> https://github.com/bw1129/PIDtoolbox/releases/tag/v0.39
>
> This was compiled with the latest (2019) [...]

Works on MacOS Catalina!

I removed all older MATLAB runtimes first, then installed using the MyInstaller_web.app. The binary that is copied to the Applications folder will not run as its missing the blackbox_decode, and bb_decode need to be opened once before the app.

bw1129, 2020-01-01:
this issue is now resolved with v0.39, so I will close it

closed 2020-01-01 by bw1129