bw1129/PIDtoolbox #30: Unable to load files from desktop

ResinSmoker, 2019-12-03:
When ever i try to dump a log to desktop and import it into Toolbox, i get an error saying that it doesn't have the required permissions. (Even in admin comparability mode)

However if i dump the files into a folder held on the Desktop, I can load / process it as normal.

bw1129, 2019-12-03:
Is blackbox_decode sitting on the desktop?

ResinSmoker, 2019-12-03:
Far as i know of the BB decode is with the Toolbox files. (Own folder)

Just find it strange that it refuses to open a file from the desktop but will work from any other folder or attached drive.

bw1129, 2019-12-03:
Bb decode MUST be where ever you open a logfile. If you’re trying to open a logfile from the desktop, it won’t work without bbdecode right there. Just copy paste it there.

ResinSmoker, 2019-12-03:
I got that... But it works literally everywhere else. Which makes no sense.

bw1129, 2019-12-03:
“Everywhere else” has to include bbdecode, as far as I know. Double check. You may have copied and pasted a folder with that in it each time.

closed 2023-05-11 by bw1129