bw1129/PIDtoolbox #31: Function Request

ResinSmoker, 2019-12-06:
Would it be possible to add the capability to process multiple files at once, then dump the resulting step-response to a GIF file?

Example: https://youtu.be/IaulQLG__eA

Spent the last two days making that and I'm really close to the ideal Servo PID's.

Thanks in advance!

-RS

bw1129, 2019-12-06:
Do you mean you would like Ptb to compute a single step response from data from several logfiles?

ResinSmoker, 2019-12-06:
Well that would be very cool too but what I was trying for was being able
to run multiple flights at differing PID values. Then be able to see all of
the logged data as a Step-Response GIF or video. This would streamline the
tuning process by allowing someone to make small changes overtime and then
see the overall results.

ResinSmoker, 2019-12-06:
Either that or have the ideal PID's calculated using data from multiple
logs.

bw1129, 2019-12-06:
Yes, that would be cool. It would be a bit cumbersome to program at the moment, just because of the way Ptb is structured, but certainly a great idea. If one could for example just go out and collect 5-10 logfiles, with various PIDs, it would be nice for Ptb to produce a complete function describing the changes in the step resp as a function of PIDs.

ResinSmoker, 2019-12-06:
Yeah, thats exactly what I've been doing with my Tricopter tuning.
https://www.youtube.com/watch?time_continue=61&v=KszUlSj1dPc&feature=emb_logo
While this flies pretty good, the tail-wag is ever present...

ResinSmoker, 2019-12-06:
Sorry, wanted to link this as well...  https://rotorbuilds.com/build/18949

bw1129, 2019-12-06:
Cool! Makes sense that it’s a small copter. You’d certainly want a light tail so the servo could move very quickly. It really has to be limited in its resp time to change directions though. This is why I would presume that it’s be better controlled from I gain the way BF currently recommends yaw tuning. While yaw authority would be high for a tri copter, the change in yaw direction would takes some time, so stopping a fast yaw maneuver would be too slow for p gain to be very effective.

ResinSmoker, 2019-12-06:
Well Yaw-I doesn't work the same on a Tricopter, simply because Quads use
differential torque, while Tri's much thrust vector. Thus a Servo is
completely driven by Yaw-P, while Yaw-I acts more to help raise it's
response to "1" then to stabilize the copters strait-line flight. Yaw-D can
be used to a limited degree but often can cause response problems due to
the added loop / filtering delay compounding the Servos slow response rate.
Keep in mind that our worst ESC's are 10-20 times faster then any Servo on
the market.

closed 2023-05-11 by bw1129