bw1129/PIDtoolbox #33: PIDtoolbox hanging, not responding, on startup splash screen,

PowerFPV, 2020-01-10:
Was working fine, and then after a computer restart the program has seemingly broke.

Upon trying to launch PIDtoolbox, the splash screen opens, a windows error sound plays, then it freezes and goes to "Program not responding", before the whole thing exits. No changes have been made between computer restarts so I see no reason as to why it doesn't launch. I'm not sure if there is anyway I can get crash/error messages.

bw1129, 2020-01-10:
Which version of Ptb?

PowerFPV, 2020-01-10:
V0.39

bw1129, 2020-01-10:
Mac? Win?

PowerFPV, 2020-01-10:
Windows 10, I've also tried reinstalling MATLAB through the included
installer and as the standalone runtime, both methods of installation work
with v0.38

bw1129, 2020-01-10:
I just downloaded it and ran it on my win10 machine, no issues, just to confirm there is no bug in that version, so it has to be something specific to your machine. Try copy paste an older executable in place of that one. you dont need to reinstall anything to try different versions. just replace the main PIDtoolbox.exe with an older version and see how that goes

pulquero, 2020-07-11:
This has just happened to me too. Was running fine, and now it only gets as far as the splash screen before being marked as not responding and then killed.

pulquero, 2020-07-11:
I got v0.391 working again by installing v0.38 over the top, checking that works, and then v0.391 over the top of that.

bw1129, 2020-07-11:
You’re just reinstalling runtime. Not Ptb. Anyway if it works that’s great. I know that 391 is based on a later version of math lab and it was designed for Carolina users. But I thought it would also work on older version of Mac. This could be the issue what version of Mac do you have.

pulquero, 2020-07-11:
This is on Windows 10.

ReefsnRidges, 2021-05-12:
> I got v0.391 working again by installing v0.38 over the top, checking that works, and then v0.391 over the top of that.

This worked for me as well... yesterday.  This morning, started pc running windows 10 and same thing.  Love the program however, sure wish this could be remedied.  Any help would be appreciated.

bw1129, 2021-05-12:
I’m not sure I understand what you mean about installing “over top” because the only installation is the Matlab runtime which should only ever have to be done once. Any new version of PTB is just a process of replacing an existing PTB program file with the latest. No installation, just delete the previous for the new.  

It sounds like you are reinstalling the runtime application each time, though depending on the operating system there have been upgrades in the Runtime version, but far as I know this is only wrt Mac. 

As I understand it, Ptb hanging at the splash screen has something to do with windows updating then it sees the program foreign and tries to block it, but if you look through the “issues” here some other people have described solutions for this. I’d follow that and see what happens. It’s not a PTB issue from a programming standpoint so I’m unable to really help much with this kind of issue.

djomaster, 2022-02-20:
I don't know what happened but since 2 days PTB 0.51 doesn't work no more. I tried on 3 different computer. I got the splash creen then nothing

funneld, 2022-08-13:
Just clearing the %TEMP% folder seems to have solved this issue on my win 10 machine.

drew-man, 2022-08-24:
> Just clearing the %TEMP% folder seems to have solved this issue on my win 10 machine.

This is what worked for me as well.

closed 2023-05-11 by bw1129

JackalFrank, 2024-03-07:
I Am having the same issue. every time im trying to open a Black box file the progam hangs at please wait.
[image]
-----------------------------------------------------------------------------------------------------------------------------
This is what i am getting from the PTBLog:

PtbVersion =

    'v0.65'

CompType =

    'PCWIN64'

Matlab_Version =

    '9.5.0.944444 (R2018b)'

currentDate = 

  datetime

   2024-03-08

Index exceeds the number of array elements (0).

Error in PTviewerUIcontrol (line 70)

Error using waitforallfiguresclosed (line 9)
Error while evaluating UIControl Callback.
------------------------------------------------------------
I have uninstalled and reinstalled and still get the same response. i have even tryed loading previously working Files from another Quad with the same results.

I checked my TEMP folder is compleetely empty.

Sesijeya, 2024-03-13:
Step 1:
After you download PTB, drop the enclosed 'PIDtoolbox_v0.6x' folder and all of its contents onto your desktop. 

Step 2:
Install using the 'MyAppInstaller_web' program located in the 'installer' folder (unless you already installed version 0.6x at some point, in which case you only need to replace the PIDtoolbox executable)

Step 3:
After installation is complete, navigate to the 'main' folder in the 'PIDtoolbox_v0.6x' folder you downloaded (DO NOT run the PIDtoolbox.exe from the programs folder! Ignore that and any shortcuts to that program, you will not be using it). Instead, use the PIDtoolbox.exe program in the 'main' folder found in the 'PIDtoolbox_v0.6x' folder you downloaded on your desktop. In there is the application you will run. Double click PIDtoolbox.exe 

Step 4: 
NEVER, put log files in the PTB main folder. They will be flushed/deleted after loading. The 'main' folder is used as a temporary place where PTB copies log files, loads them, then cleans up and deletes them. 

Note: before installing the new setup uninstall old one and delete %temp% files and follow the exact steps above (works for me)

JackalFrank, 2024-03-14:
Thank you so much for the help.

this fixed it for me!

Regards
Francois