bw1129/PIDtoolbox #35: Step Response graphs incorrect at 500hz (OpenLog via UART)

grahamcropley, 2020-02-06:
Using an OpenLog SD module via a hardware UART on a Mamba F411AIO Toothpick FC.

UART set to 250000 will only log up to 500hz (Samsung Evo 32Gb HC U1)

My rates are ~1000dps, if I do full deflection flips and rolls, it says insufficient data, and no lines are shown on the step response graph for pitch/roll with default (40-500dps)... Seems to show yaw ok.

If I up the 'degrees per second' to 2000 then it shows the familiar graph (N=30+ as expected)... it's as if it's not interpreting the lower logging rate correctly, and range needs to be doubled/tripled to be included.

Blackbox Explorer loads the data correctly. But PIDToolbox's Step Response is essential!!!

grahamcropley, 2020-02-06:
[image]
[image]
[image]

Example log:
[LOG00011.BBL.TXT](https://github.com/bw1129/PIDtoolbox/files/4167922/LOG00011.BBL.TXT)

bw1129, 2020-02-06:
Dang, I never expected anyone to log at 500hz. It won’t work. It’ll take some mucking around to make Ptb deal with those. I know, that’s bad programming. Sorry bout that. Is it not possible to log at least @1k?

grahamcropley, 2020-02-06:
Nah, it seems to get corrupted on anything higher than 1khz.. it sucks.
When loading it, it only shows <0.5 seconds if it reads it at all. iirc it says something like dimensions don't match.. or must match etc.

Maybe I need to buy one of the newer I2C / SPI based loggers instead of UART.. or I should have got a FC with blackbox.. haha. - Thanks for the reply man.

bw1129, 2020-02-06:
Ah I see. Sorry bout that.

bw1129, 2020-02-06:
If you have a 1k logfile drop me a
Link. I might be able to solve that issue.

grahamcropley, 2020-02-07:
I didn't keep any from previous testing, but I'll hook it up again tomorrow and get one for you.. thanks!

grahamcropley, 2020-02-07:
Here's a 1khz blackbox log file. No fancy flying, just hovering for ~30-40 seconds.
[LOG00018.BBL.TXT](https://github.com/bw1129/PIDtoolbox/files/4172529/LOG00018.BBL.TXT)

Seems pretty unusable in anything... I think it just can't handle it.

Blackbox Explorer only seems to show the first 200 samples. 
[image]

PIDToolBox error is 'Matrix dimensions must agree'.
[image]

Blackbox_decode output is...
[image]

bw1129, 2020-02-07:
it actually does not contain data. here is the log in BBE. there's nothing in it. that's likely an issue with the logger 
[image]

grahamcropley, 2020-02-07:
There's ~900Kb of something, but yeah I'm seeing the same, and I agree, likely all totally corrupt beyond the first buffer length... i.e. maybe too much data trying to be pushed over the baud-rate of the port.. or too much for the openlog chip itself to buffer and write to SD.

If anything it would be in blackbox_decode to see if it could skip incomplete records and look for next start of row flag (if there is one, not looked at the BBL file format).

Thanks for looking anyway, I think I can get what I need by just changing the dps up to 2000 or more.

Cheers dude

bw1129, 2020-02-07:
Ya a few bytes are used to generate header and such. But there’s no real data in there.

JoeriCapens, 2020-03-06:
Hi @grahamcropley , you may want to try setting your uart to 500000 baud by using the command line, see my comments here: https://github.com/betaflight/betaflight-configurator/commit/74445a4b4b62e3f86ebb5bc4cc37f903149d22d3

grahamcropley, 2020-03-29:
@JoeriCapens thank you!!.. sorry for the late response, but I just wanted to follow up and say it worked.. Thanks for commenting... it's much appreciated. (y)

closed 2020-03-29 by grahamcropley