bw1129/PIDtoolbox #36: Mirrored graph data in windows 10 virtual machine

jimmybe123, 2020-02-14:
just having an issue running pidtoolbox on my Linux system running a windows 10 virtual machine on virtualbox. All the buttons etc are fine but any graph data is both upside down and reversed.  Im trying to avoid doing a bare metal windows install as the only thing I want it for is pidtoolbox lol.

jimmybe123, 2020-02-14:
[image]

bw1129, 2020-02-14:
That’s the most bizarre thing I’ve ever seen. Nothing I can do on my end. It’s like a weird OpenGL issue, but nothing to do with pidtoolbox. Sorry bout that.

[image]<https://can01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F58280058%2F74496930-b31ce680-4f38-11ea-8669-ea09eb09c8c8.png&data=02%7C01%7Cbrian.white%40queensu.ca%7C549bfee0b03b421d7be408d7b0f81a3e%7Cd61ecb3b38b142d582c4efb2838b925c%7C1%7C0%7C637172451966295838&sdata=Ta2k2K9i3eD8Tq08dXduDg0IT5WxPi41roXfX7ak0uY%3D&reserved=0>

jimmybe123, 2020-02-14:
> That’s the most bizarre thing I’ve ever seen. Nothing I can do on my end. It’s like a weird OpenGL issue, but nothing to do with pidtoolbox. Sorry bout that.
> […](#)

No problem, if anyone else faces this issue I solved it by using VBoxVGA and disabling 3D acceleration in the virtualbox display settings. Thanks for a great piece of software.

closed 2020-02-14 by jimmybe123