bw1129/PIDtoolbox #37: Hardware LPF delay is calculated incorrectly.

DusKing1, 2020-02-16:
The method you use in `PTspecUIcontrol.m` to calculate(if I can say so) the delay of hardware digital LPF is incorrect. See in:
https://github.com/bw1129/PIDtoolbox/blob/5a83abf104ba9099cf2875f1a97265cb2f453fc0/PTspecUIcontrol.m#L246

What I mean is, **not all gyro's DLPF is 250hz.** 

For example, Bosch's BMI160 which is used by BrainFPV is running at 1600hz ODR(output data rate), By consulting the datasheet, the **cutoff is 523.9hz.**

I noticed this detail by re-watching UAVtech's [Bosch vs. TDK Gyro Comp vedio](url:https://youtu.be/wZM0z1Yz_6w). Though I think it is not a real big issue, it might cause some unnecessary misunderstanding, at least a pro racer asked me about this and it confused me for some time.

bw1129, 2020-02-16:
yes you are right. In the next version, I will get rid of that altogether because it doesn't matter

DusKing1, 2020-02-16:
Cool!

closed 2020-02-16 by DusKing1