bw1129/PIDtoolbox #38: pid toolbox subscripted assignment dimension mismatch and pid toolbox undefined function "tta" for input arguments of type "double"

Donpilkerton, 2020-03-08:
Not sure what the issue is here. I am trying to get pid toolbox working correctly again on my pc, It was sometime ago that I started getting these two errors when trying to load and run any file logs that I took.

bw1129, 2020-03-08:
Can you forward the logfile? Those errors have nothing to do with the cause. It’s just a pointer to the location where the code crashes.

Donpilkerton, 2020-03-08:
I believe it has been user error the whole time!! On my laptop I can pull the files straight from the SD card and upload and run them in PTB, however on my pc I have to upload my files to the main folder of PTB and it opens them right on up! Unless there is some other error as to why I can’t pull them from the SD card directly to PTB I think it was all me! Sorry about that!

bw1129, 2020-03-08:
Good, I’m glad it’s working

exocode, 2020-07-14:
I stumbled also over this issue: here is my logfile: 
https://drive.google.com/file/d/1yJi8qRTeF8Hg6dt2PxWiZ9Nj3uqBPYMb/view?usp=sharing
and these are my error messages: 

Version 0.391.0.0 (1)

[image]

[image]

[image]

bw1129, 2020-07-14:
This issue has been brought up before. The solution has been stated many times. To restate, please make sure the logfile you are trying to open is in a folder with blackbox_decode and that it is downstream of the Ptb main folder or else the logfile is placed right in the main folder. This is all described in the github setup wiki. I’ve stated that 99% of the time this is the issue but in fact it’s 100% of the time. I’ve not seen a case where this was not true. Please leave this issue active for others that might come across it. If all this does not work for you, I’m happy to look at the logfile a confirm what’s up. Just drop a Link to it here.

bw1129, 2020-07-14:
@exocode oops, this is a first, but I tried you're logfile and it does not work. I will look into why. 
thanks

bw1129, 2020-07-14:
@exocode this is a complicated issue I will try to get fixed over time. For now Im gonna hope it doesn't happen too often. let me just say it arose in part because there were so many tiny logfiles in there, a thing that BF does everytime you arm it makes one even if you dont record. Usually Ptb throws out the duds, but for some reason it got confused here.

N4V1G4T0R, 2020-07-28:
Same experience here. The probable cause is, that for some reason BF made a tiny log at the beginning. I attached the file, in case you need testing material.
[btfl_all_360_bf_2_slider.zip](https://github.com/bw1129/PIDtoolbox/files/4987177/btfl_all_360_bf_2_slider.zip)

bw1129, 2020-07-28:
The error message often has nothing to do with the cause of the crash because it appears on the stage following the actual issue. This error is 99% of the time a simple loading error because blackbox decode was not in the same folder as the logfile. PTimport.m therefore fails and the program crashes right after when it’s trying to put the looptime data into variable ‘tta’ and finds no data. The solution is to ensure bbdecode is in the folder with the logfile to its able to convert the bbl file into a csv containing the data.

N4V1G4T0R, 2020-07-28:
Same procedure as usual. And it usually works. ;-) Just with some specific files it does not (see attachment in the post before). bb_decode is in the same folder as the log. Both are in the main folder. It only happens like 1 out of 20. So, don't worry. But some hardening would be great. I would love to just open the log file where it is and not have to copy it into the main folder.

bw1129, 2020-07-28:
Ah yes, I remember this. Can’t put my finger on that issue but it tends to arise when there are too many null logfiles in there. Best to erase the flash before starting recording. There’s a really tricky part of the importing where I gather special header infor directly from the bbl files themselves. The rest is
Encrypted so I get it from the csv. The problem arises when trying to match the proper header with the proper csv. When there’s a ton of duds in there, seems the probability of an error here goes up.

IVData, 2020-08-03:
Same issue here on Windows and INAV 2.5.1:

`Error in PTload Message: Index exceeds matrix dimensions.`

`Error in PTprocess Message: Undefined function 'tta' for input arguments of type 'double'.`

[LOG00003.TXT](https://github.com/bw1129/PIDtoolbox/files/5014017/LOG00003.TXT)

bw1129, 2020-08-03:
Couple of things you can try:
1) make sure “blackbox_decode.exe” is in the same folder as the logfile you are trying to open (not a sub folder, the exact folder)

2) erase flash drive frequently. It builds up a bunch of empty logfiles, and Ptb has trouble aligning the correct header with the appropriate logfile. What happens is BF creates a logfile everytime you arm the FC, even if you don’t intentionally record a logfile. Best to delete all these before recording a logfile.

IVData, 2020-08-03:
Sorry to say, but i have done both of those things. The log is in the same "main" folder along with PID toolbox and the bundled blackbox_decode.exe. And the SD card on this craft had recently been cleared (it only has 3 logs on it, of about 30 seconds of flight each).

bw1129, 2020-08-03:
@IVData just looked at the log file. I see the issue. It's something that I've been runing bandaid solution on over the past year in the import function 'PTimport.m'. it's a horrible piece of script. I'll try to make this more universal, though it may take some time. it's just sooo bad.

closed 2023-05-11 by bw1129