bw1129/PIDtoolbox #39: Support for iNAV logs

swampmonster, 2020-03-27:
First of,
Big fan of the work you've been doing on this project. Been using it a lot with betaflight with great results!

On to my question/feature request, would it be possible to add support for iNAV logs?
From what i can tell the main issue seems to be that the naming of fields doesn't match up 1 to 1 with betaflight. I sadly dont have a matlab dev environment to help with any code but i'd be happy to provide log files.
Another possibility would be if you can point me to the fields pidtoolbox are expecting to find and me writing a tool to convert inav logs to a format that works.

bw1129, 2020-03-27:
Thanks for the request. Definitely interested in this. Been thinking of reaching out to someone in the inav group. If I could get some logfiles sent my way I’m happy to work on this. I’m not familiar with the details of inav, only to say I realize it has more sophisticated long range and multi sensor control features, which would also be nice to implement into Ptb. Besides Ptb current functionality, what aspects of inav would be valuable for analysis in Ptb? Maybe I could continue this discussion on a Main inav Facebook group where I could probably also get some logfiles.

swampmonster, 2020-03-27:
Logs i'd be happy to provide but they do have GPS data in them so i would prefer if i can email them or otherwise send privatly.

bw1129, 2020-03-27:
Yes, good point. Maybe we can link up over Facebook Messenger. You can find me on the Blackbox log review facebook page.

bw1129, 2020-05-01:
So I’ve been making some progress on this but it hit a roadblock. For some reason Ptb can’t accurately compute set point. This is most likely because it does not understand INAV rate system. I’ve posted the question on inav Facebook site but no responses. Question: what are the main parameters? Eg in BF it’s i)rc_rates (main), ii)superrates (creates expo but modulates max rates) iii)rc_expo (pure expo function with no affect on max rates). I presume inav does not have superrates, so may have to dig into src code to see how this is computed.

swampmonster, 2020-05-10:
Here's what i was able to dig out from the cli and a screenshot of how it looks in the configurator.

```
roll_rate = 60
Allowed range: 6 - 180

pitch_rate = 60
Allowed range: 6 - 180

yaw_rate = 50
Allowed range: 2 - 180

manual_roll_rate = 100
Allowed range: 0 - 100

manual_pitch_rate = 100
Allowed range: 0 - 100

manual_yaw_rate = 100
Allowed range: 0 - 100

rc_expo = 70
Allowed range: 0 - 100

rc_yaw_expo = 20
Allowed range: 0 - 100
```

[image]

bw1129, 2020-05-10:
Hey thanks! I’ll start looking into this again. I might have to visit the RC set point code in inav. Seems it’s don’t different from BF.

wx4cb, 2021-06-08:
> I honestly didn’t think there were any INAV users that would be interested in this level of tuning. I ended up simplifying the loading code because it was getting so cumbersome that I had to ignore INAV. INAV uses different variable names. I could look into making it work again if there seems to be some more community demand for it. Do you have any sense how many INAVers actually use this tool?

to be honest, i only just found this tool as it was mentioned in  JBs live stream today so i figured i would try it out seeing as the wiki said it supported inav 👍 

to be honest i don't think a lot of people know about this in that world. maybe if you're on fb you can go ask in the inav fixed wing group and take a look there ....

wx4cb, 2021-06-08:
for me, it would be more about looking for noise than anything else.

bw1129, 2021-06-08:
Ok, in that case version 0.392 will still work. V0.4 is just for sophisticated step response tuning

charlesmarseille, 2021-07-01:
Hi guys, I'm also looking at getting PT working with iNAV. Just wen through the code quickly and indeed the variable names are different, and feedforward is called axisRate instead of axisF. I think if we could just replace those and remove unwanted columns from the decoded file it could be read by PT.

robmulally, 2022-01-10:
I didn't want to make a new issue and saw this was open but Inav 4.0 user here  all my Quads on iNav for Position Hold etc , I also noticed there is some heavy lifters moving to Inav as well . 0.45 seemed to work fine 0.5 seems to crash a lot, Wish i could provide reasons but it gives a crash sound with no window or error message. Just wanted to add my support for other iNav users.

bw1129, 2022-01-10:
Yea there are some known bugs. One common one is make sure there are no spaces in the logfile name (nor weird characters).

closed 2023-05-11 by bw1129