bw1129/PIDtoolbox #40: Error in PTload - Invalid file identifier. Use fopen to generate a valid file identifier

jasc76, 2020-04-12:
Hi

I get this error. Already did a reinstallation. 
BF4.1 Log
https://1drv.ms/u/s!AqdVRzkS_NXbg-cmh335qhfCOOEcRQ?e=hhRHQi

bw1129, 2020-04-12:
no issue with loading it here. Make sure blackbox_decode is in the same folder as that logfile.

[image]

jasc76, 2020-04-12:
it is in the folder of the log .... :-(
I tried deinstalling pidt and mathlab but didtn help 

[image]

bw1129, 2020-04-12:
Do you have any logfiles that work? If so, put this logfile in the location of the working logfiles. Double confirm that there are other working logfiles in that folder. This is definitely a directory issue. Are you using v0.39?

bw1129, 2020-04-12:
If all that still fails put the logfile and bbdecode in the same folder as Ptb executable.

jasc76, 2020-04-12:
no, i wont load any log
trying blackbox_decode from cmd works fine

[image]

putting both (bb log and bb decode) in the ptb application folder results in the same error

bw1129, 2020-04-12:
Has it ever worked? If not, I’d reinstall. Make sure you use the supplied installer. Don’t go to the matlab site yourself and download runtime. In most cases that won’t work. Use the supplied installer. Btw, is this windows or Mac?

jasc76, 2020-04-12:
yes it worked some time ago

I deinstalled ptb and matlab, afterwards deleted all folders under program files and reinstalled with the ptb installer (including new matlab download)

its windows

jasc76, 2020-04-12:
Its kinda funny
if I run it from the installation folder I get the error

If I run it from the download folder, it works

I copied everything from the donwload over to the programfiles application folder and it doesnt work

bw1129, 2020-04-12:
Ok, so problem solved. Yea I should have mentioned, leave everything where it is and work from there. You should be able to put that entire download folder somewhere else as long as you leave everything in place. Just use a shortcut to the executable if you want a startup icon in your desktop. Just to confirm you’re using a mac right?

jasc76, 2020-04-13:
Its windows...
Solved, well, kind of. It doesnt work if I run it from the installation directory: C:\Program Files\PIDtoolbox\application
It works if I run it from here: C:\Users\jansc\OneDrive\Quadcopter\tools\PIDtoolbox\PIDtoolbox_v0.39_win\PIDtoolbox_v0.39_win\main

it even does not work if I copy everything from the working folder to the folder in ProgramFiles

bw1129, 2020-04-13:
the only thing installed is runtime. The pid toolbox executable is kinda independent. But it should work pretty much anywhere else on your main C drive. It might have something to do with user rights within certain folders. Not sure. Also it’s worth noting there’s a little txt file that reads the previous log location. It’s possible this is messing with the directory if you have another copy of that somewhere. Try looking for that file in other locations of your computer. It would have been created in the selected folders only

ai91, 2020-05-08:
Confirming - getting same issue when installed in **Program Files**.
Problem seems to be in necessaty to create **logfileDir.txt** file.
Workarounds:
1. move PIDtoolbox to not protected path
2. start as admin
3. modify access permission on **Program Files\PIDtoolbox\application** - grant full access to all users

Topher84, 2021-07-06:
> Confirming - getting same issue when installed in **Program Files**.
> Problem seems to be in necessaty to create **logfileDir.txt** and intermediate ***.csv** files.
> Workarounds:
> 
> 1. move PIDtoolbox to not protected path
> 2. start as admin
> 3. modify access permission on **Program Files\PIDtoolbox\application** - grant full access to all users

Can confirm that right-clicking and running as admin fixes it for me. I did not have to do anything else. I used the installer to install the most recent release and had the same issue.

bw1129, 2021-07-07:
Thx for the info. Please leave this here for others that may experience similar issue.

closed 2023-05-11 by bw1129