bw1129/PIDtoolbox #43: Cannot load logfile: Error in PTload

PeterBenko, 2020-05-22:
PTB fails to load the attached log with the following console output:
```
peter.benko@Peters-MacBook-Pro-3 application % ./run_PIDtoolbox.sh /Applications/MATLAB/MATLAB_Runtime
------------------------------------------
Setting up environment variables
---
DYLD_LIBRARY_PATH is .:/Applications/MATLAB/MATLAB_Runtime/runtime/maci64:/Applications/MATLAB/MATLAB_Runtime/bin/maci64:/Applications/MATLAB/MATLAB_Runtime/sys/os/maci64
dyld: Library not loaded: @rpath/libmwlaunchermain.dylib
  Referenced from: /Applications/PIDtoolbox/application/./PIDtoolbox.app/Contents/MacOS/PIDtoolbox
  Reason: image not found
./run_PIDtoolbox.sh: line 29: 59902 Abort trap: 6           "./PIDtoolbox.app/Contents/MacOS/PIDtoolbox"
peter.benko@Peters-MacBook-Pro-3 application % ./run_PIDtoolbox.sh /Applications/MATLAB/MATLAB_Runtime/v97 
------------------------------------------
Setting up environment variables
---
DYLD_LIBRARY_PATH is .:/Applications/MATLAB/MATLAB_Runtime/v97/runtime/maci64:/Applications/MATLAB/MATLAB_Runtime/v97/bin/maci64:/Applications/MATLAB/MATLAB_Runtime/v97/sys/os/maci64
Error in PTload 

 Message:
Index exceeds the number of array elements (0).
Error in PTprocess 

 Message:
Undefined function 'tta' for input arguments of type 'double'.
```

Dimensions of arrays being concatenated are not consistent. Granted, I haven't gotten PTB running on my setup yet at all so this might be a user error. Am I doing something wrong or is my BF just too old (3.2.2). Could it be a mac issue? I'm on Catalina.

[LOG00012.TXT](https://github.com/bw1129/PIDtoolbox/files/4668555/LOG00012.TXT)

bw1129, 2020-05-22:
Ah I see the problem. in older versions of BF rc_rates was called "rc_rate" with a single number for all axes, and Ptb cant find the string so it leaves it empty but needs it, and crashes in PTimport,m. I can modify this to deal with that issue, but the best option is to upgrade to at least BF3.5. If you have an older FC, there are performance editions of 3.5. That said, I'll put it on the to-do list.

PeterBenko, 2020-05-22:
I'm sadly bound to versions released by ImmersionRC for my Vortex150, since it doesn't play nicely with the public releases :(

Is there an option for me to post-process the log in a way Ptb could read it?

bw1129, 2020-05-22:
Ah darn. That’s too bad. You might be able to if you can edit a portion of the header where it shows rc_rate 100. You need three 100s in a row I think, or whatever your rates are. Not sure why it only has two “100”s. Can’t remember how the rate system was setup in older BF. Same for expo I think.

bw1129, 2020-05-23:
@PeterBenko I am working on a version of Ptb that will also read older logfiles. It works with one exception. You have to log at least at 1k (1000kHz). That log is recorded at 250Hz and you can't do meaningful spectral analysis at that rate. If you could send another logfile sampled at 1k, that'd be great. I think it's a simple fix then.

PeterBenko, 2020-05-23:
Oh that's great news! I've been struggling with manually editing the log but that seems to already fail already in blackbox_decode. I'd guess they do version specific parsing. 

I can provide further logs really soon - with one caveat. The one provided should have been logged at 500hz at 1khz pid loop clock (and I think 2khz gyro sampling rate), due to otherwise having quite heavy logging drops even at 250000 baud rate on OpenLog. I should have some logs at hand in an hour or so.

PeterBenko, 2020-05-23:
Attached are some logs with 2khz gyro sampling rate, 1khz pid loop an 1khz logging. The headers seem to be a bit messed up but they look surprisingly _okay_ in the Blackbox explorer.

[LOG00026.TXT](https://github.com/bw1129/PIDtoolbox/files/4672030/LOG00026.TXT)
[LOG00027.TXT](https://github.com/bw1129/PIDtoolbox/files/4672031/LOG00027.TXT)
[LOG00028.TXT](https://github.com/bw1129/PIDtoolbox/files/4672032/LOG00028.TXT)

bw1129, 2020-05-23:
you are still only logging at 0.5kHz. I am also noticing broken logfiles. your SD card cant keep up

bw1129, 2020-05-23:
[image]

bw1129, 2020-05-23:
NOTE looptime 500. That needs to be 1000.   [image]

PeterBenko, 2020-05-23:
Hmm yes I get that too in the explorer but then either that number is a relative value (gyro vs pid) or one of the systems is showing wrong data. Here's screenshots from BF: 
[image]
[image]

PeterBenko, 2020-05-23:
I've only now noticed my SD card is on the ["poor performers" list (SandiskUltra 32 class 10)](https://github.com/betaflight/betaflight/blob/master/docs/Blackbox.md#microsdhc-cards-known-to-have-poor-performance)" in BF docs. All I can do is report back when I get a better one.

bw1129, 2020-05-23:
Maybe the wrong configurator for that version. Go into CLI and get the denom variables set up. CLI is always right

PeterBenko, 2020-05-23:
```
# get deno
gyro_sync_denom = 4
Allowed range: 1 - 32

pid_process_denom = 2
Allowed range: 1 - 16
```
If I understood the docs right, the calculations are:
gyro:
(max rate = 8khz)/gyro_sync_denom = gyro clock
8khz/4 = **2khz**
and pid loop:
(gyro sample rate = 2khz)/pid_process_denom = pid clock
2khz/2=**1khz**

I have to admit I fail to get the BB rate via CLI

bw1129, 2020-05-23:
Interesting. Surprising it’s giving you 500 in both BBE and Ptb. Wrt Ptb I can confirm lograte because I do a diff on the sample times and get ~2000microsec, so it’s definitely 0.5kHz lograte. One thing you could try is setting both loop and lograte to 2k and see what happens... be worth a shot. Could be a bug in an old version of BF. I thing v3.2 had several revisions, like a 3.2.5 even I think.

bw1129, 2020-05-23:
yes, if the "500" looptime in BBE means 500us (I was thinking 500Hz). But the cycletime recorded in the log looks like this: 
Here are the first ten entries of column 2 [cycletime in us] of one of the logfiles. 

diff(dataA.DataMain(1:10,2))
ans =
        4041
        4040
        4049
        4059
        4046
        4049
        4034
        4048
        4038
        4043

PeterBenko, 2020-05-23:
Sooo I stole a card from my GoPro that's ~5x faster than the previous one but I guess that didn't help the log quality very much
[LOG00029.TXT](https://github.com/bw1129/PIDtoolbox/files/4672280/LOG00029.TXT)

I played around on the bench a bit, so the following logs are nonsense but maybe you could confirm a faster logging rate at least? I set logging to 2khz and am curious if the logs reflect that at all.
[LOG00030.TXT](https://github.com/bw1129/PIDtoolbox/files/4672281/LOG00030.TXT)
[LOG00031.TXT](https://github.com/bw1129/PIDtoolbox/files/4672282/LOG00031.TXT)

bw1129, 2020-05-23:
I'll take a look

bw1129, 2020-05-23:
ah, whatever you did in LOG00029.TXT worked. I think you were at 2k gyro sampling 2k pidloop 1k logging there. 
gyro_sync_denom=2
pid_process_denom=2
with solid ~1000us cycle time
diff(tta(1:10))

ans =

        1032
        1011
        1025
        1022
        1035
        1030
        1016
        1016
        1018
LOG00030.TXT worked ok but had tons of errors and was trying to achieve 2k (500us cycle time), but missed a lot
diff(ttb(1:10))

ans =

         506
         506
         563
         511
         503
         504
       29306
         510
       96594
gyro_sync_denom=4
pid_process_denom=1

PeterBenko, 2020-05-23:
I think the logs got a tiny bit messed up but let me get one last live flight log before total nightfall then, but I'm glad it worked. 30 should be indeed live footage at 2k/2k/2k.

bw1129, 2020-05-23:
LOG00031_2.TXT doesnt work (it's 4/1)
only LOG00029.TXT worked. (2/2)

PeterBenko, 2020-05-23:
Just to avoid any confusion 29 has 3 sessions:
Session (of 3) |  gyro denom | pid denom | gyro frequency | pid frequency | error % reported by blackbox_decode
--- | --- | --- | --- | --- | --- |
1 | 2 | 2 | 4khz | 2khz | 0.27%
2 | 2 | 2 | 4khz | 2khz | 0.24%
3 | 4 | 1 | 2khz | 2khz | 97.27%

This is confusing to me tho since I'd assume a slower sampling clock leaves more time for logging.
Which one do you mean?

bw1129, 2020-05-23:
it was 29, file 2 that worked, and from the header info it's 2/2:
{'gyro_sync_denom'          }    {'2'                                      }
{'pid_process_denom'        }    {'2'                                      }

PeterBenko, 2020-06-01:
So I think I've gathered some useful logs with a flashed OpenLog and a better card
[LOG00047.TXT](https://github.com/bw1129/PIDtoolbox/files/4712682/LOG00047.TXT)
[LOG00048.TXT](https://github.com/bw1129/PIDtoolbox/files/4712683/LOG00048.TXT)
[LOG00050.TXT](https://github.com/bw1129/PIDtoolbox/files/4712684/LOG00050.TXT)

bw1129, 2020-06-01:
Those logfiles are recorded at 500hz (a sample every 2ms). It has to be recorded at 1000-2000Hz. Most common today is logging at 1000hz. Im not sure why you are unable to do that. Im assuming the configurator is giving you bogus numbers.

PeterBenko, 2020-06-01:
😠 This is kinda annoying, the lograte was supposed to be synced with the pidloop. Is tehre any way for me to determine the _actual_ rate before bothering you?

bw1129, 2020-06-01:
I noticed when opening in Black Box explore that you can look in the set up tab and it shows Luke time at 500 which I assume is 500 Hz. Another way is to look at a trace in explorer and use the Mark tool it’s not very easy to see but I think you can see that with that with the smoothing off you can see the steps of the samples

bw1129, 2020-06-01:
one thing you can do is quick tests on the bench with props off. you dont need to be in the field to get it right. just play around until you figure out what combination results in 1k log rate. at that point, I already think Ptb will work.

PeterBenko, 2020-06-01:
I think the 500 looptime is us, which represents the gyro looptime. 

However, measuring the time between log entries does seem to result in 500hz intervals between different data points. Strange, since the log entry counter at the bottom (next to time) seems to count up at ~500hz rate, but counts +2 between data points resulting in a log# that is "synced"  to ms. 

Thanks, I can work with this, it would seem there actually is a bug in BF, as odd as I find that.

bw1129, 2020-06-01:
Ok. So 2k gyro sampling rate. Then you use denom 4 for pid loop rate (which divides that by 4), which equals 500hz.

bw1129, 2020-06-01:
if unsure, zoom in x and y scale (x time zoom is at top of screen, y zoom in plot settings). then use 'mark' tool to confirm time between samples. must be 1ms or >. Here it's 2ms (500hz logging).
[image]

PeterBenko, 2020-06-01:
Yeah I'm pretty sure the blackbox logging rate input is just bugged. Example: this was at 4khz gyro/2khz pid and !!2khz lograte!!.
[LOG00088.TXT](https://github.com/bw1129/PIDtoolbox/files/4713349/LOG00088.TXT)
[image] -> Looks effectively a 1khz lograte

Whereas this is at 4khz gyrp/2khz pid and 1khz lograte. 
[image]
[LOG00091.TXT](https://github.com/bw1129/PIDtoolbox/files/4713471/LOG00091.TXT) -> effectively 500hz lograte.

I think something somewhere is just messed up.
But at least I have acceptable settings now. Thanks!

PeterBenko, 2020-06-01:
Is your patch for older BFs merged already?

bw1129, 2020-06-02:
It’s not merged yet but I can get you a version and send directly when I get a chance.

bw1129, 2020-06-02:
I can compile you a mac version. you're on mac right?

PeterBenko, 2020-06-07:
I have access to both Win and Mac but I have a Mac with me in the field so that'd be preferred, yes :)

closed 2023-05-11 by bw1129