bw1129/PIDtoolbox #44: Crash on MacOS Catalina - latest version from PIDtoolbox

mrRobot62, 2020-06-21:
Hi folks,

I installed the latest mac zip file from PIDtoolbox on my macbook pro. After installing und starting the program, I get below crash dump

It seems there is something wrong with the math lab libs. 

Any ideas what I can do?

Bernd
...
Time Awake Since Boot: 110000 seconds
Time Since Wake:       9600 seconds

System Integrity Protection: enabled

Crashed Thread:        0

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    DYLD, [0x4] Symbol missing

Application Specific Information:
dyld: launch, loading dependent libraries
DYLD_LIBRARY_PATH=/Applications/MATLAB/MATLAB_Runtime/v95/runtime/maci64:/Applications/MATLAB/MATLAB_Runtime/v95/sys/os/maci64:/Applications/MATLAB/MATLAB_Runtime/v95/bin/maci64:/Applications/MATLAB/MATLAB_Runtime/v95/bin

Dyld Error Message:
  Symbol not found: _mecab_get_feature
  Referenced from: /System/Library/PrivateFrameworks/CoreNLP.framework/Versions/A/CoreNLP
  Expected in: /Applications/MATLAB/*/libmecab.dylib
 in /System/Library/PrivateFrameworks/CoreNLP.framework/Versions/A/CoreNLP

bw1129, 2020-06-21:
Was it running ok on the previous version or is this your first install?

mrRobot62, 2020-06-21:
honestely I'm not sure, I used PIDtools a couple of month ago, and I installed several things on my mac again - as PIDtoolbox - it seems, that currently two different MatLab libs were installed v95 and v97 and I do not know, why on above crash dump the v95 is used.

This crash dump is created, if I start PIDtools out of my /Application folder. So maybe, I should copy the newest app (0.3.9.1) manually to this folder. I will do this

bw1129, 2020-06-21:
Delete old versions and reinstall

closed 2020-06-25 by bw1129