bw1129/PIDtoolbox #45: Can not load log file (maybe same as #43)

mrRobot62, 2020-06-22:
Hi,
I use latest BF 4.1 FW on my copter, running under MACOSX. Few minutes ago I made a short flight and tried to load log file into Ptb - unfortunately with some error messages during loading.

[image]
[image]

If useful, I can upload my bbl file too.

With BB-Explorer it's possible to load this file

bw1129, 2020-06-22:
Sure, drop the logfile here and I’ll take a look.

mrRobot62, 2020-06-22:
Hi brain, thanks fortgesponnenes.log should attaches in my thread - isn’t it? 🤔

Viele Grüße
Bernd

mrRobot62, 2020-06-25:
Any news, why I can‘t load my logfile into PIDtoolbox?

bw1129, 2020-06-25:
Quick question, have you recently upgraded from a previously working version?

bw1129, 2020-06-25:
@mrRobot62 I have no issue opening your logfile here. Make sure you select logfile 2 in there. logfile 1 is <2sec long. 99% of the time the reason people can not open a logfile is because of a directory issue. Make sure blackbox_decode is in the same folder as the logfile. Because BF makes logfiles the same name it's easy to get confused here, sometimes grabbing what looks like the correct logfile, or else Ptb gets confused as to which directory it's supposed to look, based on a pre-existing version of Ptb with another "main" folder with bb decode and more importantly the logfiledir.txt file causing Ptb to look in the wrong place. I always recommend putting all logfiles in the same place in the main folder with Ptb and bb decode, and just renaming the logfiles to avoid duplicates. also, when upgrading Ptb, just grab the new Ptb app/exe and replace the old one with it, in the same folder. Let me know if any of this helps.

mrRobot62, 2020-06-25:
Hi Brain, thanks for response. Problem solved. You're right, blackbox_decode (on MACOSX too) must be available in the same folder as the log file(s).

PTB Application can be located at any place on the harddiks. Normally every applications will be installed under /Application folder on a MAC and that works.

Problem is blackbox_decode, this 90k little file must be in the same folder as the log file :-)

Brain, maybe I'm wrong, but if I take a look into PTImport.m, you call via system-call blackbox_decode on row 41 with parameter filename. Do this filename contain the file path too?

If not, maybe it's possible to include file path + file name and set it as parameter to the decode. If this work, than the decode-app can be located "anywhere"

Maybe it'S a good solution to work every time with absolute path & filenames and not a "change dir" into a "working folder".

bw1129, 2020-06-25:
The reason bb decode has to be in same folder is not just for it’s own sake but it generates a csv file and better to have that located in same folder as the bbl for extracting data and working with. I wouldn’t want to deal with csv data being in one place and the logfile in another place.

mrRobot62, 2020-06-25:
ah - ok - I was wondering "who" create this tmp folder with a CSV file inside - so it's bb_decode. Ah - ok.

BTW: I know (a little bit) MatLab. What do I need to compile your code and build an executable package? I assume it's the compiler-package from them, isn't it? Is to compiler package in the HOME-Edition included?

bw1129, 2020-06-25:
If you have Matlab you can run the scripts out of the command window. But you will need a bunch of toolboxes and they’re costly unless you’re affiliated with a university of something.

mrRobot62, 2020-06-25:
unfortunately not :-( - but maybe via my daughter she studies in Berlin. I close this issue, because it's solved

closed 2020-06-25 by mrRobot62

mrRobot62, 2020-06-30:
Hi Brian,

Unfortunately I have no longer connections to university.

BTW:
What do you think, make it sense to have a German wiki pages for PID-Toolbox on Git?
If yes, I can try to start with a translation. One big advantage is, I will learn a lot ;-)

Regards
Bernd

bw1129, 2020-06-30:
Sure, if you’re u for the challenge.

bw1129, 2020-06-30:
One minor issue is the wiki is pretty dated. The basics should still apply though I suppose

mrRobot62, 2020-07-01:
Hi, ah ok, mhm, first step I start with a translation and afterwards we (you) include more and more updates.

 

Question is, how, we can choose languages - via link or is there an other solution on git? Not sure, mhm...

 

Gesendet: Dienstag, 30. Juni 2020 um 21:23 Uhr

mrRobot62, 2020-07-01:
Because my idea is to work with different sites and not one big monster ;-) - like today.

 

I will think about it .....

Gesendet: Dienstag, 30. Juni 2020 um 21:23 Uhr

mrRobot62, 2020-07-01:
I see, I have no access rights to update the wiki

 
 

Gesendet: Dienstag, 30. Juni 2020 um 21:23 Uhr

bw1129, 2020-07-01:
You could fork, make edits, then submit as issue or pull request. Not sure how best to do it. I don’t wanna open editing to public.

mrRobot62, 2020-07-03:
No prob - will fork it.

Let‘s what the future will bring ;-)

Nice weekend

bw1129, 2020-07-03:
Great! Ya I was just looking and it seems you can’t do a pull request for
The wiki, so it wouldn’t track your contributions. It may turn out to be best to just drop a Link in my wiki to your forked wiki, maybe, or I could just add it to Ptb and credit you in the wiki if you’re ok with that.

mrRobot62, 2020-07-04:
Hi Brian,

yeah, will start with translation next days - and if I have a „first release“ you can share the link as supposed

Nice weekend (here in North-Germany bad weather to fly :-( - rain since a couple of day)

Bernd

www.myyl.com/balance

mrRobot62, 2020-07-04:
BTW:

I don’t want to  translate 1:1 because (because that’s easy and in my opinion not my goal) I think, sometimes it’s a little better to explain the sense behind the english text - supported by some additional explanations - ok for you?

Not everyone has an expert level on tuning ;-)

Bernd

www.myyl.com/balance

bw1129, 2020-07-04:
Sounds good.
We should continue communication another way. Can you look me up on the BB Facebook group? I’m Brian White.

mrRobot62, 2020-07-04:
Hi ho Brian,

will search you on FB :-)

Have a look onto : 
https://github.com/mrRobot62/PIDtoolbox/blob/master/README.md <https://github.com/mrRobot62/PIDtoolbox/blob/master/README.md>

First edition (currently nothing more :-O ) -maybe with a couple of typos :-/ 

Viele Grüße
Bernd

www.myyl.com/balance

> Am 04.07.2020 um 12:47 schrieb Brian White <notifications@github.com>:
> 
> Brian White.