bw1129/PIDtoolbox #46: PTload error. The second argument must be a message character vector...

exocode, 2020-06-22:
I have this error:
`blackbox_decode error Please ensure your log files are in the main...`

`PTload error. The second argument must be a message character vector or a handle to an existing waitbar`

`Error in PTprocess Undefined function 'tta' for input arguments of type 'double'`

[BTFL_BLACKBOX_LOG_GepRC_20200622_220328.BBL.zip](https://github.com/bw1129/PIDtoolbox/files/4815436/BTFL_BLACKBOX_LOG_GepRC_20200622_220328.BBL.zip)

I found that when I moved that file into the App folder it worked. Did the installation process changed? Because the README says, that I do not have to move that file...

bw1129, 2020-06-22:
The log file must be in a folder with bb_decode program. Confirm that. Otherwise I think that folder just has to be at or downstream of the main folder. Also make sure you don’t have two versions of Ptb on your computer. Always overwrite the old with the new pidtoolbox app/exe.

bw1129, 2020-06-25:
I have no issue opening your logfile. 99% of the time the reason people can not open a logfile is because of a directory issue. Make sure blackbox_decode is in the same folder as the logfile. I always recommend putting all logfiles in the same place in the main folder with Ptb and bb decode, and just renaming the logfiles to avoid duplicates. also, when upgrading Ptb, just grab the new Ptb app/exe and replace the old one with it, in the same folder. Let me know if any of this helps.[image]

closed 2023-05-11 by bw1129