bw1129/PIDtoolbox #47: Couldn't install latest version-Mac 10.13.6

Droniac75, 2020-07-05:
I couldn't install the latest version.  The MyAppInstaller_web would crash every time.

bw1129, 2020-07-05:
If you have a previous version, you don’t need to use the installer. Just replace the Ptb app/exe

Droniac75, 2020-07-05:
I finally got v0.38 MyAppinstaller_web to install.  Go the same error as orlovskiy in issue 17.  Finished the install.  Double click on the PIDtoolbox icon to run app, now I get this error dbx: 

Cannot locate a valid install area 
applauncher cannot locate a valid install area.  If a MATLAB Runtime is not in the default location, then you must establish a valid install area by using system preferences panel.

bw1129, 2020-07-05:
Check to see if contents are in the apps folder. Otherwise make sure you have it installed where you have admin rights. I’ve left similar issues up there because they show solutions so also look at those

Droniac75, 2020-07-05:
[image]

bw1129, 2020-07-06:
Not Matlab runtime. You should have a separate folder for PIDtoolbox.

bw1129, 2020-07-06:
oh, sorry I see the Ptb folder as well. Looks like you placed it there though. I just install it from the desktop and it automatically created a PIDtoolbox folder there. But you should be able to place logfiles in any normal folder from your desktop, as long as BB decode is in there. Something is up with the way you installed that.

mrRobot62, 2020-07-08:
Hi @Droniac75 I'm a Mac-User too, is your problem solved yet?

Droniac75, 2020-07-08:
@mrRobot62 unfortunately, no

mrRobot62, 2020-07-09:
ok, hope I'm able to help you. I assume u installed PIDToolbox out of the box from package and you have 
a) PIDToolbox-App installed in /Application folder 
b) a folder on your hard disk names like PIDToolbox0.391 (or something else)
c) u have installed MATLab runtime v95

As I understand your screenshot, this folder is stored under /application - right?

Ok, you can try following steps:

a) move folder out from /application to an other place on your hard disk
b) remove in /Application all PIDToolbox apps (if available)
c) out from copied PIDToolbox folder copy ONLY the PIDToolbox application into /Application
d) open a terminal and go to your PIDToolboxfolder
e) run the shell script run_PIDToobox.sh 

Does Ptb starts now? Please note: on MAC there is NO splash screen. On my MacBook Pro start takes approx 45secs (be patience ;-) 

If not, please copy & paste the log output in this thread (output from script)

if it works with the shell script close Ptb, than try the same with double click of Ptb out of /Applications

Hope this approach is more ore less clear written :-)

good luck

mrRobot62, 2020-07-09:
don't ask - i do not know why it is posted in bold :-O

bw1129, 2020-07-09:
it's too bad it's so complicated for the occasional mac user. the thing is, most dont have this issue, so I can't help but think it's realted to the manner in which it is being installed. All I do is drop the main and install folders on the desktop and double click the installer. is there something unique about how one is installing that is related to this failure I wonder?

closed 2023-05-11 by bw1129