bw1129/PIDtoolbox #51: Can't open anymore PIDToolbox on Windows 10

AndroidGX, 2020-11-16:
Hello,

PIDToolbox was working like a charm but recently for no reason, I can only see the splashscreen for few seconds and then nothing more.
I tried the following without any success :
- Trying to fully uninstall/reinstall both pidtoolbox and matlab 93
- Trying to update to matlab 9.9 (not recognized by pidtoolbox) then revert back to 93
- Verifying right paths to environment variables (everything was OK already)
- Rebooting many times, clearing cache, various optimisations, without any success

Any idea ?
Many thanks in advance

bw1129, 2020-11-16:
If it was working fine then suddenly stopped running, it’s most likely a change on your Windows machine, possibly an update or change in user rights. Nothing about PTB changed. Wish I could help more but this is not a Ptb issue

AndroidGX, 2020-11-16:
Probably due to a windows update, yup, but then it could be interesting to be compatible again with the latest version of Windows 10 update, since users like me can't use it anymore then :-(

bw1129, 2020-11-16:
It might not be an issue of
Comparability but user rights. Windows is probably trying to block the program because it doesn’t recognize it. I would start there. The executable was created on a windows machine that’s up to date and should run on any machine. Now that you’ve reinstalled the incorrect version of runtime it’s not going to work either. You should uninstall everything and make sure all traces of PTB are deleted. Then reboot the machine then try a full reinstall that includes using the myinstaller. Don’t try installing runtime yourself from the website. That won’t work because it has to be exactly the right version in which it was compiled and that is controlled by “myinstaller” app.

AndroidGX, 2020-11-16:
> You should uninstall everything and make sure all traces of PTB are deleted. Then reboot the machine then try a full reinstall that includes using the myinstaller.

Already tried that, unfortunately not working.
Tried also to run with admin privileges but same issue.

bw1129, 2020-11-16:
I can try recompiling a version based on the latest Window's updates. I could work with you directly later this morning over a zoom call if you want. I might be able to figure out what's going on face to face. where are you located? maybe PM me on Facebook. Im on the BB Facebook group.

AndroidGX, 2020-11-16:
Ok that would greatly appreciated, as your tool is very useful, many thanks, I'll try to catch you in facebook 👍

AndroidGX, 2020-11-16:
Tried to contact you via FB (not sure if it's you), let me know whatever you need for debugging

AndroidGX, 2020-11-17:
Ok as seen via Facebook, I replaced a old binary (exe) from previous version (3.9 for example) and it actually launched the app, then I tried to replace with latest executable 0.392 and it worked back ! strangely.. not sure why, but glad your recommended trick worked well in my case 👍 
Many thanks for the help, much appreciated 🥇

bw1129, 2020-11-17:
That’s great! Sounds like a weird Windows thing

flosean, 2021-01-04:
Also meet this problem, and try replace 0.39 exe file to 0.392 and it works again. On latest windows(20H2).

bolsunovskyi, 2021-04-03:
Same issue :( It was so not in time. I was in a field without internet. Going to tune my PIDs but failed. The solutions works for me too.

AndroidGX, 2021-04-03:
I formatted my computer recently and fully reinstalled Windows 10 from scratch and never encountered the issue since for now 👍 
Cheers'

closed 2021-04-03 by AndroidGX