bw1129/PIDtoolbox #52: Unable to open blackblox log in PIDToolbox on MAC OSX

c0debreaker, 2021-01-30:
First, I tried exported the blackbox log to csv. However, in PIDToolbox - Log Viewer, it is not able to open the csv file. It is greyed out and I can't select it.

So I tried opening the blackbox log directly. After I clicked on Load+Run, it failed with the following errors

[image]

[image]

[image]

I'm not sure what I'm doing wrong. Any help would be greatly appreciated.

c0debreaker, 2021-01-30:
I also tried renaming the .csv file to .txt and retried the same process. I got the same errors like above.

c0debreaker, 2021-01-30:
I now tried it on Linux. I got this

[image]
[image]

I waited for 1.5 hrs and import progress bar is still like this
[image]

Am I doing something wrong? I can read my blackbox log successfully using Blackbox Explorer.

Unfortunately, I don't have a Windoze machine.

bw1129, 2021-01-30:
PTB does not support opening csv files anymore. Just use the original .bbl file and make sure you put the file in the same folder as the blacbox_decode program file. It’s all in the wiki. Follow it closely. Best to follow the blackbox log Facebook group as well. UAVtech’s blackbox site

closed 2021-02-01 by c0debreaker