bw1129/PIDtoolbox #54: Please bring back D-Term prefiltered

FritzWlan, 2021-06-08:
Please bring back D-Term prefiltered view to the spectrum analyzer. I use it a lot so fine-tune the D-Term filters. Really need it.

bw1129, 2021-06-08:
Not sure I’ll be bringing that back. The scaling was flawed in previous versions. I usually recommended people not use it. It’s not part of BF variables but was computed in PTB. I’ve heard in the grapevine that it might actually become a part of BF variables output to logfiles, but don’t quote me. I will therefore wait till BF 4.3 release to see what happens there.

FritzWlan, 2021-06-09:
Not pointing fingers at anyone, but since the usual lack of negotiation between the most contributing devs goes up I doubt BF 4.3 will be released anytime soon and I highly doubt D-Term prefiltered will be a blackbox debugging option. I know one has to take the computed D-Term prefiltered signal with a grain of salt but it is really helpful in many situations, especially when tuning D-Term Lowpass filters or deciding if a static notch is more helpful on gyro or just on D-term. Having to use two versions of PID-Toolbox is quite a pain.

bw1129, 2021-06-09:
I’ll likely bring it back in next version. It has be right though. Will need to use ones D gain setting to adjust the scaling. Otherwise the spectrograms are giving you useless false info. You’d be just as well to rescale the gyro to bring up the noise to see how it might look since it’s largely just an amplified version of that. Anyway, I’ll get it right

closed 2021-08-16 by FritzWlan