bw1129/PIDtoolbox #55: 0.3x and 0.4x - Renaming log files

Zuldan, 2021-06-09:
Is there any chance you could adjust the code to not rename log files to remove the spaces in the name? Or at least rename the log files back to what they were before opening them on exit?

It’s killing my OCD for having nicely named and ordered logs for each of my quads ;-)

bw1129, 2021-06-09:
It’s a pain in my end. Doesn’t run with spaces. Bad practice.

bw1129, 2021-06-09:
One solution would be to use underscore instead of a space.

Zuldan, 2021-06-10:
An underscore would be a step up from spaces I guess. Is it not possible to rename the file back to what it was before exiting the program?

Actually...how about not messing with the original log file at all. Would it be better practice to copy the log file to say C:\Windows\Temp\PIDtoolbox or C:\Users\<username>\AppData\Local\Temp\PIDtoolbox and opening it from there? then the original file is always left intact. It’s also a good place to extract the temporary CSV files.

bw1129, 2021-06-10:
Copying and pasting files is just a recipe for bugs and other issue. File names are only changed if it sees a space. I rather not mess with this and risk injecting new unforeseen issues into the program. The best practice is for the user not to use spaces and just replace them with an underscore. Then Ptb will not change anything.

Zuldan, 2021-06-11:
No worries. I’ll use underscores. Appreciate your response:)

closed 2021-06-11 by Zuldan