bw1129/PIDtoolbox #56: 0.4 / 0.42 hangs on creating spetral analyzer chart

jasc76, 2021-06-17:
Hi
tested with .4 and .42

I loaded 5 logs and that works fine. Only thing I notice is that the mouse coursor still shows processing (this circle) but opening and analysis of step response works.
Opening the spectral analyser is also fine but on "Run" he starts processing but hangs infinite. I also notice a windows notice "bing" maybe from some background process
[image]

bw1129, 2021-06-17:
If you hear a “ding” sound that means it crashed. Try restarting. If using logfiles with multiple logfiles in a single file, this can be problematic. I always recommend single individual files. To make these, unplug the battery the plug back in. Then enable BB log switch, arm, then take off. If you arm then quickly disarm then arm again and fly you create extra very short files and the more of these the more issues Ptb has in linking the correct extracted csv file with the correct header file in the bbl file. When this happens the new version may still continue running but may not have the file properly loaded. There’s still some issues I need to resolve there to properly deal with this. Anyway, if you have a particular logfile that does weird things it’d be awesome f you could forward it to me. It can be really helpful in detecting and solving these issues that seem to have always plagued Ptb. Thanks 🙏 !

jasc76, 2021-06-17:
Thanks
found it. The first log file is too short or defective. I included it here.
If I omit the file and just use the other 4, it works. 
[btfl_001.zip](https://github.com/bw1129/PIDtoolbox/files/6673114/btfl_001.zip)

bw1129, 2021-06-17:
Perfect!

closed 2023-05-11 by bw1129