bw1129/PIDtoolbox #57: pidtoolbox to right clicm menu / open .bbl from everywhere

jasc76, 2021-06-19:
Hi

I organize my builds and corresponding logs and diff backups. It would be great to be able to open pidtoolbox with right click on any bbl file. Is this possible? If not, would it be possible to get rid of the restriction to have bbldecode within the folder of the logfile?

bw1129, 2021-06-19:
bb_decode, well, decodes the bbl file. I suppose in windows you might be able to put it in the system folder so that it can be called from anywhere but I don’t have any other solution.

jasc76, 2021-06-20:
> 
> bb_decode, well, decodes the bbl file. I suppose in windows you might be able to put it in the system folder so that it can be called from anywhere but I don’t have any other solution.

Ah, works like a charm ;-) putting decode in a folder from the path variable works... Oh man, I should have know that :-) I'm into computers long enough

bw1129, 2021-06-20:
That’s awesome! Please leave this issue post because others may find it useful

closed 2023-05-11 by bw1129