bw1129/PIDtoolbox #58: Error in PTload

AndreySemjonov, 2021-06-24:
For EmuFlight logs on latest 0.42 PIDtoolbox getting error
[image]
Do not have same issue with 0.392

Here log:
[8.zip](https://github.com/bw1129/PIDtoolbox/files/6708055/8.zip)

If this is issue on EMUF side than we can fix it.

bw1129, 2021-06-24:
I see the error. Dmin and FF strings empty. Fixed and will be part of next release. 
[image]

bw1129, 2021-06-24:
ah, here's a version with the fix. Go ahead and try this one. 
https://www.dropbox.com/s/n2njf6ys47g5fky/main.zip?dl=0

AndreySemjonov, 2021-06-24:
Thanks. It is OK now

bw1129, 2021-06-24:
Excellent!

Keep this 'issue' here so others can download the fix if needed.

massimilianodossi, 2021-06-30:
hi , can you provide the corrected file for MAC also ?

bw1129, 2021-06-30:
No prob. I’ll post it here in a bit. New release in the works as well soon

bw1129, 2021-06-30:
emu fix for mac users. you also get a few extra fixes for free with this version ;-)
https://www.dropbox.com/s/d357g4klnhe5wyy/main_forMac.zip?dl=0

massimilianodossi, 2021-06-30:
sorry but the ptload error persist

bw1129, 2021-06-30:
I just ran it using the emu log posted above and it works fine. you must have a different issue. normally this doesnt matter, but delete the old PIDtoolbox.app then try the logfile above (linked in the first post at beginning of this issue)

charlesmarseille, 2021-07-01:
while trying to open an iNAV log, I get a new error (was getting the one posted above before):
Error in PTload
unrecognized variable name 'setpoint_0_'.

So I guess you fixed the variable names, but not for the "special" iNAV ones. maybe a simple fix?

bw1129, 2021-07-01:
PTB 0.4+ went through complete overhaul. The file importing functions had to be rewritten to accommodate the new multi-file structure. As a result I had to make a decision how complicated I wanted it to be and I opted for a very simplified approach. Compare PTload and PTimport now compared to v0.392 for example. So I don’t rename anything now. It comes is as Blackbox_decode exports it and that’s all determined by the main firmware developer. Both BF and EMU use the same basic variable names for set point etc, but INAV doesn’t. So given the small INAV user base I decided to stop support for INAV. I made a post in INAV asking who uses PTB and got like 3-4 responses. PTB v0.392 still works for INAV users so that’s where I left it. If more INAV users expressed interest I’d work at it.

charlesmarseille, 2021-07-01:
Yes, I saw your ask for the interest in iNAV users, not so many responses. I then tried 0.392, and got the same error. Just by changing these 2 header variables I got it working though.
Thanks for your great work, really enjoy that visualisation of the data.

Quick-Flash, 2021-07-01:
Hey whatever it takes to help you keep supporting emu we would be happy to help.

bw1129, 2021-07-02:
Hey thanks Kevin! So far so good. I think as long as the main variable names come out of bb_decode the same we’re generally good to go.

closed 2023-05-11 by bw1129