bw1129/PIDtoolbox #59: Error in PTload

nishanthe1, 2021-07-15:
[image]

When i try to put a BB log it shows that message.

bw1129, 2021-07-15:
link to logfile please. also take a look at responses to similar issue posted by others

nishanthe1, 2021-07-15:
Hi, 

Sorry I am new to this, but it says this type of file is not supported. 
[image]

And I had a look at other threads but I couldn't find any similar issues.

nishanthe1, 2021-07-15:
[btfl_002.zip](https://github.com/bw1129/PIDtoolbox/files/6825171/btfl_002.zip)
Here you go, sorry i managed to get it to work.

bw1129, 2021-07-15:
Great! Be great if you could post here what you did. Valuable for others

nishanthe1, 2021-07-16:
The blackbox Log was corrupted so therefore it was not reading the log correctly.

closed 2021-07-16 by nishanthe1