bw1129/PIDtoolbox #60: 0.44 Spectral Analyzer not working.

FritzWlan, 2021-08-05:
Win10Pro, PIDToolbox 0.43 worked fine. Erased 0.43 and unzipped 0.44. Stepresonse works fine. Spectral Analyzer just gives me a windows chime (error sound) and nothing happens. Maybe you can look into that, Thanks!

bw1129, 2021-08-05:
I just tried it on my windows machine and it worked fine. I recompiled and uploaded again. Could you try downloading the win version again and giving it a try?

drjing, 2021-08-06:
Hi, I have the same problem as well, upgraded from 0.42 and running windows 10 pro. I am installing using the web installer. I will try just copy the main files over and see if it works, report back later.

drjing, 2021-08-06:
I have solved the windows chime by just copy all files from the main folder to the installed location.

closed 2021-08-16 by FritzWlan