bw1129/PIDtoolbox #61: ArduPilot Support

tenten8401, 2021-08-06:
Any chance we could see ArduPilot support in the future? This tool would be super useful to tune large UAVs as autotune in ardupilot does not work very well.

Logging format is standard and there are many tools capable of decoding it that could be used for reference if needed.

bw1129, 2021-08-06:
It’s certainly possible. If you have a logfile I’d be happy to take a look. It’d be critical that set point was included in the logfile.

LimitlessGreen, 2021-08-10:
For example this one:
[ardu_log.zip](https://github.com/bw1129/PIDtoolbox/files/6964832/ardu_log.zip)

https://github.com/ArduPilot/pymavlink could be useful, it did some of the work already.

bw1129, 2021-08-10:
Thanks! I’ll look into it.

bw1129, 2021-08-10:
the only challenge for me is extracting the data and importing it into Matlab. I don't know what to do with a .bin data file. be nice if there was a tool to convert the data to a csv. Let me know if something like this exists.

tenten8401, 2021-08-10:
Mission Planner is able to convert .bin to .log which is a CSV file, it's in bottom left panel you'll have to click a few arrows over to see dataflash logs section

tenten8401, 2021-08-10:
Looking at the .log I don't know how useful it'd be, can't tell if setpoint is involved. Only looking at it in notepad though.

bw1129, 2021-08-10:
Could you convert that file to log file for me. I could work from there.

tenten8401, 2021-08-10:
Here you go:
[ardu_log (2).zip](https://github.com/bw1129/PIDtoolbox/files/6965015/ardu_log.2.zip)

bw1129, 2021-08-11:
just looking at the data, it's a bit cumbersome but I can manage to plot a few things. Unfortunately I dont know exactly what Im plotting. There's 'RATE' and 'IMU', 'IMU2', 'RCIN' etc, but not sure what corresponds to 'Gyro'. I think most of this stuff is typically in auto level mode so it's hard to say whether it'd work using the methods in PTB.

tenten8401, 2021-08-11:
I'll see if I can get an acro log tomorrow and find a way to convert it to CSV with setpoint and gyro in there.

tenten8401, 2021-08-11:
I know the .bin file has the data as I can plot it with https://plot.ardupilot.org
[image]

mishaturnbull, 2021-08-31:
Also interested in this topic, and can generate some logs to test if needed (although not acro...not that confident in my drone yet :).  What I can tell you is this:

* ATT.DesRoll and ATT.Roll may not be the best lines to analyze for ardupilot PID tuning (likewise for pitch and yaw).  AP uses a rate PID, which drives the angular rotation rate rather than position, and that's what typically is tuned in the parameters.
* You can find those values in the RATE message, with R and RDes corresponding to roll and roll desired.
* Raw gyro data gets non-trivial, since different hardware has different numbers of gyros.  On a pixhawk cube, with IMUs, the gyro messages are all under the IMU1/2/3 messages (GyrX, GyrY, and GyrZ).  There isn't a dedicated GYRO message; linear accelerometers and gyroscopes are combined into one message.
* The .bin files can be programmatically parsed with the DFReader class in PyMavlink; I have a version without the full dependency hierarchy [here](https://github.com/mishaturnbull/TrashBin/blob/master/src/logutils/DFReader.py)
* You can pretty easily generate .csv files from .bin files via either Mission Planner or a few lines of Python utilizing that DFReader class, although I've never written a line of Matlab so I have no idea how much effort it would take to get a .m file
* If it's useful, you can also (if the parameter is enabled) log the P, I, and D individual contributions to the PID loops.  Those show up under PIDR, PIDP, PIDY for roll/pitch/yaw, along with the target rate (setpoint) and the actual rate.

tenten8401, 2021-10-30:
Just wanted to put this link in here as it's got a useful table on it with some info about the log files: https://ardupilot.org/copter/docs/common-downloading-and-analyzing-data-logs-in-mission-planner.html

tenten8401, 2021-11-08:
It looks like we also may need IMU_RAW checked in LOG_BITMASK parameter if we need to log

AndKe, 2022-08-17:
Dataflashlogs contain huge amounts of data, and are in a binary format to ease storage/reduce write operations.
As for expanding dataflashlogs (.bin)  :
If you install MAVProxy (could be as easy as "pip install mavproxy") 
then you will have a tool "mavlogdump.py"  - that can quickly dump all, or only the kind of data you are interested in.
Feel free to request logs when you get around to looking at ArduPilot.

km5es, 2022-08-19:
I am interested in this also. There currently is no tool to analyze PID step responses in ArduPilot. I can provide logs in csv format with requested data if it helps.

malinhiles, 2022-09-10:
Just wanted to chime in here as I was recently curious if ptb could handle ardupilot log files as well. @bw1129 I noticed this "Create MATLAB File" Button in mission planner. Not sure if it suits your purposes, but none the less I wanted to chime-in in case it serves helpful in the future.

closed 2023-05-11 by bw1129

tenten8401, 2023-05-11:
Was this implemented in the new version or just closed? :(

-Isaac

Closed #61<https://github.com/bw1129/PIDtoolbox/issues/61> as completed.

bw1129, 2023-05-11:
Just cleaning house. No idea if when ardu