bw1129/PIDtoolbox #62: Version 0.44 - StepResponse - several clicks on RUN create new entries in graphs

mrRobot62, 2021-08-07:
maybe it's a bug. 

Load two files and run a step response analysis
click run
Graph is build - for both files - ok
P,I,D,Dm,F list two entries
peak - graph two points
latency - two points
click run a again
graph with two new colors
P,I,D,Dm,F list four entries
peak - graph four points
latency - four points
click run a again
graph with two new colors
P,I,D,Dm,F list six entries
peak - graph SIX points
latency - FOUR points
a.s.o

Bug is repeatable

bw1129, 2021-08-07:
Not a bug. It’s designed to keep adding plots. If you select same files to plot each time it just keeps adding same data only with different colours

mrRobot62, 2021-08-07:
ah - ok - make sense, I know the feature but I thought if nothing changed inside file list nothing happens if user click on RUN

but it's not a problem :-) - thanks for this quick response

closed 2021-08-07 by mrRobot62