bw1129/PIDtoolbox #63: KISS functionality

ethanxy, 2021-08-13:
Please add KISS log analysis function.
40 secs flying data from FETtec FC recorded by openlog blackbox in the link below for your reference.
[Google Drive](https://drive.google.com/drive/folders/1Q2h4Ydzs7UaYaibXRIJG_2MkSFADAinC?usp=sharing)

Thanks!

spatzengr, 2021-08-13:
@ethanxy , the problem with KISS logging is the Setpoint is not calculated correctly -- or called out correctly in the header file.  @bw1129 , I'm working with Alex - slowly - to resolve.  He cut new HEX files to fix, but it needs to be tested.  I need to flash to my KISS board and hook it up to test it out.  Or maybe I can get Krunked to make some logs.

spatzengr, 2021-08-13:
Krunked said he will talk with Alex and make some logs to vet it out.  I'll pass along.

ethanxy, 2021-08-13:
Thank you.
Let us know if more tester are required.

bw1129, 2021-08-13:
Perfect! Thanks

closed 2023-05-11 by bw1129