bw1129/PIDtoolbox #64: Bug? StepResponse tool - Latency values different between Function-Graph and Latency-Graph

mrRobot62, 2021-08-14:
Not sure if this is a bug - maybe a misunderstanding on my side ?

As I understand latency this is the time in ms between 0 and the time on x-Axis if graph cross the Y-1-Line - right?

**Example**: (Function-Graph)
[image]
Data-Tips : for X 22ms

**But on Latencygraph**
[image]
On this graph I get a latency from around 12.6

Why do I see this differences?

** Question 2** - I do not understand Peak time(ms)

bw1129, 2021-08-15:
No, it’s the latency at half height (0.5 mark). I chose that because it’s a closer estimate to the xcorr lag between set point and gyro. Also it avoids the issue of curves that don’t hit the target (1) line, or do so later in the trace but doesn’t really make sense then. It also resembles a kind of median latency in a cumulative distribution.

mrRobot62, 2021-08-15:
ah, good to know - thanks for your response.

BTW: I'm working on the German-PIDTool-Wiki based on 0.44 and rewrite it completely (more or less). It will published on [Betaflight.de](www.betaflight.de). If done, I close the old PIDTool-Wiki on github because it's obsolete.

Are there some other secrets, which I should know for the wiki :-) ?

Kind of a median latency is a good approach but for users with no knowledge behind the implemented math, it's not bad to know .

closed 2023-05-11 by bw1129