bw1129/PIDtoolbox #65: Cannot open INAV blackbox log

avsaase, 2021-08-16:
I am not able to open the attached INAV blackbox log using 0.44:
[image]

The file opens just fine in INAV Blackbox Explorer 3.4.0  and decodes without issues (0.1% dropped frames) with INAV's `blackbox_decode` 0.4.6.

The logging frequency is ~250Hz (2KHz PID frequency and 1/8 logging rate). I am aware this is probably not enough to make use of all features of PIDtoolbox but I figured it should be able to open the file.

[LOG017.TXT](https://github.com/bw1129/PIDtoolbox/files/6994142/LOG017.TXT)

bw1129, 2021-08-16:
Make sure blackbox_decode_INAV is in folder with the logfiles. Log rate 1k minimum.

avsaase, 2021-08-16:
Yes, I have `blackbox_decode_INAV` in the same folder as the log file. 

Can you confirm it is the logging rate that causes this? I think in principle it should be possible to do some step response analysis at 250Hz. Probably not ideal but I can't go higher with a serial logger because INAV logs a lot of variables and does not currently support selecting which variables are logged.

bw1129, 2021-08-16:
no issue loading it here. just wont work on the tools at 250hz. crashes in both spec analyzer and step resp when trying to plot
[image]

bw1129, 2021-08-16:
ok, managed to get it running. I can fix this. maybe I could upload a new version to github. I'll just replace v0.44 with this
[image]

avsaase, 2021-08-16:
That would be awesome. Thanks! And how about the spectrum analyzer? Blackbox explorer shows the spectrum up to half the logging rate. Even at 250Hz, this could provide some useful information.

I had some problems updating to 0.44. Maybe that caused my import problem. I'll try removing and reinstalling.

Again, thanks a lot!

bw1129, 2021-08-16:
yes, spec still works. just only goes to 125Hz (because of nyquist limit!)

bw1129, 2021-08-18:
@avsaase I made that edit on the win version so it should work now with logs < 1k. I didnt issue it as new release because it was so minor, so just download the latest executable and replace with the one you have now. I'll do same for mac and linux in a minute

avsaase, 2021-08-18:
Do you know what could the error I posted in the OP? I replaced the executable (and the other files) with the updated 0.44 Windows version but I still get the same error. I have Matlab installed on my system. Could that have something to do with it?

bw1129, 2021-08-18:
Double confirm blackbox_decode_INAV is in there. Make sure there is write access to folder. It could be read only.

bw1129, 2021-08-18:
Also make sure logfiledir.txt is in pidtoolbox main folder. This is a file that records and holds previous selected folder location so when you reselect it redirects you to that folder. If all that doesn’t work drop the file right there in main folder and see if that works as a test.

avsaase, 2021-08-18:
> If all that doesn’t work drop the file right there in main folder and see if that works as a test.

Okay, this works. Before, the LOG file was in a folder with write access, along with blackbox_decode_INAV.exe. I checked if it was because the LOG file was on a different drive than the PIDtoolbox executable but that was not the problem.

bw1129, 2021-08-18:
Ok so you have a solution. Not an ideal solution, but a solution nonetheless.

avsaase, 2021-08-18:
That's true. Thanks a lot for helping me out. I understand you don't want to do a new release right now, but I think more people could benefit from blackbox analysis at lower logging rates.

bw1129, 2021-08-18:
Yes the step tool for sure! I’m impressed it worked so well. That’s promising

bw1129, 2021-08-19:
@avsaase I injected a small bug in the program doing this. I ave to revert for now and try to solve it correctly. It's ok for your purpose for now, but I need to fix this

avsaase, 2021-08-19:
Sure, no problem. I have a working executable now.

Regarding my error opening log, if I open the first log by first placing it in the same folder as PIDtoolbox, then subsequent logs can be opened from any folder.

avsaase, 2021-08-23:
Will the ability to open <1kHz logs be part of the next release? If so then this issue can be closed.

bw1129, 2021-08-23:
Yes. It’ll be in next release. Thx

closed 2021-08-23 by avsaase

avsaase, 2021-08-24:
This is technically a new feature request but it would be great if PIDtoolbox could import decoded CSV logs. 

Right now, with logs that contain multiple flights, you need to decode the log from PIDtoolbox, select which flight you want (after which it deletes the decoded CSV files) and repeat for the other flights from the same file.

bw1129, 2021-08-24:
PTB does import decoded csv files technically but I think I understand your request. Others have asked same. I’ve been thinking about how to do this as it’s a bit tricky but should have a solution soon

avsaase, 2021-08-24:
The option to import multiple flights from the same file in one go, with an indication of which log is which flight, would achieve the same thing for me.