bw1129/PIDtoolbox #68: v0.44: Spectral Analyzer Window doesn't open

hberg539, 2021-09-08:
Hi,

i'm using Windows and trhe v0.42 worked without issues. With the new v0.44, i am having issues opening the Spectral Analyzer window with the same log files.
When i click on it, i just get the windows "alert sound", like a message window opens (but none does):
[image]

Here is the log file i use:
[btfl_001.zip](https://github.com/bw1129/PIDtoolbox/files/7129165/btfl_001.zip)

I'm using the betaflight 4.2.

bw1129, 2021-09-08:
Try downloading it again. There was a fix and re upload so you might just have the older version with the bug

hberg539, 2021-09-08:
Now it works.
Thank you for the great tool!

closed 2021-09-08 by hberg539