bw1129/PIDtoolbox #69: 0.44 Clicking on spectral analyzer does nothing

sohamsp14, 2021-09-08:
When I click on spectral analyzer, nothing happens. There's a "ding" sound and that's it. Nothing loads

bw1129, 2021-09-08:
Download latest version again. You got a buggy version that was since replaced. Please leave this issue. Several people are saying same thing then they solve it but delete the issue and it keeps coming back

FritzWlan, 2021-09-12:
I don't know if maybe a windows update messed something up, but I get the same error like the thread opener. When clicking anything in the spectral analyzer I get the windows error chime and nothing happens. Downloading the recent version again does not help.

bw1129, 2021-09-12:
I think there is a bug where once you use the heatmaps and go back it doesn’t function after that. Restart the program which clears the workspace and it should be ok. I need to fix the bug though. Working on it

sohamsp14, 2021-09-12:
Reinstalling the latest release did fix the issue. But, as others stated, the bug does appear again after u open the heatmap and close it. Clears after a restart. But as I read, it's getting fixed. 

PS: Keeping the issue open to help others with a fix.

sohamsp14, 2021-09-26:
This issue has been happening again even with the latest release. Happens even after a restart now.

bw1129, 2021-09-26:
The fix is coming soon. Just finishing a few additional things before I release

closed 2023-05-11 by bw1129