bw1129/PIDtoolbox #74: Invalid file identifier error

Zuldan, 2022-01-12:
I'm trying to open a set of logs using PIDtoolbox v0.5, however I receive the following error. The logs are from Betaflight 4.3 RC2. Not sure what the issue could be. 

[image]

[Logs.zip](https://github.com/bw1129/PIDtoolbox/files/7851836/Logs.zip)

bw1129, 2022-01-12:
Have you confirmed blackbox_decode is in the folder, and that there are no spaces in the logfile name?

Zuldan, 2022-01-12:
There are no spaces in the filenames. Whats interesting is I can load each log individually. I only get the error when I select all of them at once and load.

bw1129, 2022-01-12:
Ah, one of them is < 1sec long or even less. That happens when you quickly arm then disarm and create a file that actually has no data. See if you select a bunch but leave out the ones < 2sec (control plus mouse or apple then mouse in mac I think and you select multiple individual files)

Zuldan, 2022-01-12:
Thanks for the explanation Brian. Appreciate your help.

closed 2022-01-12 by Zuldan