bw1129/PIDtoolbox #80: Return linux builds

mituritsyn, 2022-04-09:
there are linux pidtoolbox users, please don't ignore us

ipetepete, 2022-04-17:
I managed to get it to run on my machine (Manjaro) but it throws errors when loading the log files and just hangs. Not sure what to make of the error either "Index exceeds the number of array elements. Index must not exceed 0." 

I'm only posting in the hopes that maybe we *nix-er's can help each other out.

mituritsyn, 2022-04-19:
> I managed to get it to run on my machine (Manjaro) but it throws errors when loading the log files and just hangs. Not sure what to make of the error either "Index exceeds the number of array [...]

Hi, did you use latest available version? Try to place log file in main PTB directory along with blackbox_decode file

kickfliph, 2022-05-04:
Hello there,

@ipetepete, can you please share with us how you manage to make it work on Linux?

Regards,

mituritsyn, 2022-05-04:
> Hello there,
> 
> @ipetepete, can you please share with us how you manage to make it work on Linux?
> 
> Regards,

it's pretty easy - you need to install matlab runtime
then preload paths, for my system it looks like:
$ export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/local/MATLAB/MATLAB_Runtime/v93/runtime/glnxa64/:/usr/local/MATLAB/MATLAB_Runtime/v93/bin/glnxa64/:/usr/local/MATLAB/MATLAB_Runtime/v93/sys/os/glnxa64/
afte you can just run pid toolbox
$ ./PIDtoolbox

mituritsyn, 2022-05-04:
> 

did you install runtime with pid toolbox binary? If memory serves me correctly, I downloaded it myself from the official website and installed it according to their instructions

bw1129, 2022-05-04:
The Matlab runtime, and Matlab program are different things. The runtime only allows a user to run Matlab compiled programs. Matlab itself is a proprietary academic platform / programming language for math/science/engineering and is pretty expensive, not to mention you have to pay for each toolbox with separate licences. Octave is an open source platform that functions very similar to Matlab, and can run most Matlab .m files, scripts and functions, but not all, and somethings work pretty different from Matlab, but it's the closest that exists. Anyway, I apologize for not keeping up with the linux versions, but it is a lot of extra work (many details are quite different from the others), and I am sorry but there is not a lot of incentive to keep that going as regularly as the others. Once I am really happy with the working version, and I find time, I may consider a linux version. My apologies for the inconvenience.

kickfliph, 2022-05-05:
Hello there,

@mituritsyn Thanks for that information, really appreciate it. I was able to install the Matlab runtime and export the path but got lost on what version of pidtoolbox you are using that gives 
 you the ./PIDtoolbox to run it, I have downloaded all the different OS versions but I am not clear which one works for me.

Thanks for all the help,

Regards,

mituritsyn, 2022-05-05:
@kickfliph you need PID Toolvox v0.45 it is the last vertion that support linux

mituritsyn, 2022-05-05:
@bw1129 btw I've managed to adapt major functionality under Octave

[image]

bw1129, 2022-05-05:
That’s awesome! 
There was an OpenGL issue with octave that resulted in really slow graphics. It used to be painful plotting.  Is that resolved?

mituritsyn, 2022-05-05:
> That’s awesome! There was an OpenGL issue with octave that resulted in really slow graphics. It used to be painful plotting. Is that resolved?

it was a proof of concept, i've had very little time and don't know when I'll be able to go back to refinement. all my changes are in my fork. 
there's only the basic functionality and I've achieved launching through a lot of crutches. so far the bottle-neck is in the smoothing functions. I haven't encountered any opengl problems yet.

kickfliph, 2022-05-05:
Thanks again, 

So, I just download the _0.45 version for Linux and I just try to run it and I'm getting this errro[1], so I google it and I found this[2], so my question is: should I roll back the ML-runtime version to 2013a in order to make it work in Linux or its something else I'm doing wrong?

[1] $ ./PIDtoolbox
./PIDtoolbox: error while loading shared libraries: libmwlaunchermain.so: cannot open shared object file: No such file or directory
 
[2]https://www.mathworks.com/matlabcentral/answers/267562-how-can-i-resolve-this-mcc-runtime-error-cannot-open-shared-object-library#answer_211306

Regards,

bw1129, 2022-05-05:
You have to use the same version of runtime in which the program was compiled. It’s simple as that. And to ensure you get that version of runtime, you should use the “my installer” app provided.

kickfliph, 2022-05-05:
Thanks again @bw1129,

I finally make it run, thanks again for all your help to all of you. I still have the libGL error message in the back[attached] of the running session, I have the missing library install but I don't understand why the error still showing. 

Thanks one more time, really appreciate all your support.

Regards,
[image]

Sol1du2, 2022-05-11:
> The Matlab runtime, and Matlab program are different things. The runtime only allows a user to run Matlab compiled programs. Matlab itself is a proprietary academic platform / programming language [...]

That's totally understandable. Would it be possible however, to provide some sort of guide in how to best try and compile the code in linux? Maybe a recommended Octave version so maybe some of us might be able to try and build some of the versions a bit easier.

mituritsyn, 2022-05-11:
@Sol1du2 unfortunately you need matlab license to build binaries. Octave not fully compatible with all matlab functions. You can try this project but it is on a draft stage:
https://github.com/mituritsyn/OctavePIDtoolbox

Sol1du2, 2022-05-11:
> @Sol1du2 unfortunately you need matlab license to build binaries. Octave not fully compatible with all matlab functions. You can try this project but it is on a draft stage: https://github.com/mituritsyn/OctavePIDtoolbox

Goes to show you how much I know about Matlab :laughing: Thanks for the reply though!

I'm now wondering if anyone has tried it with wine. I might give it a shot one of these days.

nerdCopter, 2022-05-19:
my run script, if it helps anyone: https://gist.github.com/nerdCopter/022709392824295846b5ce0b0992b578

bw1129, 2022-05-19:
I just compiled a linux version of PTB 0.55, it's under releases. Can someone let me know if it is working?

bw1129, 2022-05-19:
[john-solo-day4-emuf_all.bbl.zip](https://github.com/bw1129/PIDtoolbox/files/8732129/john-solo-day4-emuf_all.bbl.zip)

nerdCopter, 2022-05-19:
For completeness, posting the solution here as well:
Source: https://www.mathworks.com/matlabcentral/answers/363058-deploying-standalone-matlab-exe-on-a-linux-based-server
Last comment said rename the MATLAB file `libmwcoder_types.so` to something else.
     File location depending where you installed it -- usually in `usr/local/MATLAB/MATLAB_Runtime/v93/bin/glnxa64/`
PTB 0.55 will then work.

mituritsyn, 2022-05-20:
[image]

Throws some warnings and errors but working fine:
```
libGL error: MESA-LOADER: failed to open radeonsi: /usr/local/MATLAB/MATLAB_Runtime/v93/sys/os/glnxa64/libstdc++.so.6: version `GLIBCXX_3.4.29' not found (required by /usr/lib64/dri/radeonsi_dri.so) (search paths /usr/lib64/dri)
libGL error: failed to load driver: radeonsi
libGL error: MESA-LOADER: failed to open radeonsi: /usr/local/MATLAB/MATLAB_Runtime/v93/sys/os/glnxa64/libstdc++.so.6: version `GLIBCXX_3.4.29' not found (required by /usr/lib64/dri/radeonsi_dri.so) (search paths /usr/lib64/dri)
libGL error: failed to load driver: radeonsi
libGL error: MESA-LOADER: failed to open swrast: /usr/local/MATLAB/MATLAB_Runtime/v93/sys/os/glnxa64/libstdc++.so.6: version `GLIBCXX_3.4.29' not found (required by /usr/lib64/dri/swrast_dri.so) (search paths /usr/lib64/dri)
libGL error: failed to load driver: swrast
Warning: Integer operands are required for colon operator when used as index
> In PTSpec2d (line 23)
  In PTplotSpec2D (line 63)
  In waitforallfiguresclosed (line 9)
```

> For completeness, posting the solution here as well: Source: https://www.mathworks.com/matlabcentral/answers/363058-deploying-standalone-matlab-exe-on-a-linux-based-server Last comment said rename [...]

thanks solved spectral analyzer freeze

bw1129, 2022-05-20:
Great thanks! Yea the real reason for the psd line plot errors is that script is where the filter/set point-gyro delay is calculated from a Matlab function called ‘finddelay’. But in fact the real problem is a function that ‘finddelay’ uses called ‘xcorr’. Xcorr is a cross correlation function and it’s not working properly in the version of Matlab I have for Linux. Not surprising, it’s a bit dated (2017). Once I update that it should be fine. But for now at least there’s a working 0.55 linux version. I will also explore Ptb in octave again as well. It’s been a while since I ran it and it was pretty slow in previous versions. Will try it again soon

Rejdukien, 2022-06-22:
> For completeness, posting the solution here as well: Source: https://www.mathworks.com/matlabcentral/answers/363058-deploying-standalone-matlab-exe-on-a-linux-based-server Last comment said rename [...]

Chiming in with the same error symptoms on the spectral analyzer window, but with **Windows 10** using the provided runtime in the 0.55 release zip.
In this case, the file `%PROGRAMFILES%\MATLAB\MATLAB Runtime\v93\bin\win64\coder_types.dll` needs  to be renamed/removed.

AndKe, 2022-08-17:
Linux build, pretty-please.

closed 2023-05-11 by bw1129

AndKe, 2023-05-11:
what happened here? issue closed, last release 9hr ago has no Linux build?

bw1129, 2023-05-11:
Linux support ended long ago. Sorry

AndKe, 2023-05-11:
That's sad why? - isn't it just about compiling it for linux ?  I mean ...doesnt matlab "just does it"?   ...not to mention you promised   :cry: